cyoung/stratux #767: Standalone AHARS web page output suitable for HUD

N129BZ, 2019-02-24:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Would it be possible to add a web page with your AHARS in a non-bordered and no-title page, suitable for use on a HUD?  Your AHARS graphics are excellently done, very smooth.  I have attached an image of green-on-black of your AHARS.  If the user could select the resolution that would be fantastic. 

Brian Manlove

[image]

Helno, 2019-02-24:
@egid Would this be tough to make?

N129BZ, 2019-02-24:
I got that png by replacing the colors in stratux's css file for the AHARS display.  I can then see the whole page on a kivic HUD - if it could be shown by itself, maybe adding a vertical airspeed tape on the left and a vertical altitude tape on the right, it would rival some REALLY EXPENSIVE stuff... just a thought.

I'm not sure how stratux is generating the markup for drawing their AHARS but it's extremely smooth.  Maybe someone could point out the code for that?

Helno, 2019-02-24:
This is where I would look. 

https://github.com/cyoung/stratux/blob/master/web/plates/js/ahrs.js

N129BZ, 2019-02-25:
For reference, this is a good size for the Kivic - 780 x 455.  Thanks for the reference to that ahars.js.

[image]

N129BZ, 2019-03-11:
I have now almost completed a hud web page that is using SVG graphics with CSS and JQuery which I can run in the Raspberry Pi using the Chromium browser in kiosk mode.  That gives me a full screen without borders, menu bars, etc.  

One question I have is - How hard of a hit is it on the Stratux for me to be POSTing to the **getSituation** url at a rate of 20 hz?  That gives me a pretty smooth screen output.  I left it running overnight and it appeared to work just fine.  It makes a lot more sense to me to not have to deal with heartbeats, keepalives, etc., in this situation.  My next step is to see if I can get the assets on the Stratux itself, in which case I can just run the composite video cable right from there.

Screen shot attached.  Comments welcome.
[image]

ninelima, 2019-03-11:
From memory that sounds about the polling I used fro Kwik EFIS.

http://members.iinet.net.au/~ninelima/efis/index.html

https://gitlab.com/ninelima/kwikEFIS

for HUD (#767)

N129BZ, 2019-03-24:
I have a working prototype of Stratux HUD that I will post in my fork.  I installed the most minimal footprint possible on the Stratux RPi, including the Openbox window manager, and the Chromium browser.  I added a subdirectory at /var/www/hud/ and put the content html, js, img, css directories and files there.  

The window manager is loaded and Chromium is launched in Kiosk mode on the RPi to the URL of **http://localhost/hud/hud.html**, the Kivic hud unit plugs directly into the 3.5mm composite video jack on the RPi.

This reduces the "overall footprint" of adding a HUD display to the package, because it only requires the video cable and the Kivic HUD.  Some of the other adaptations require a 2nd RPi, which means more cables, more USB ports, etc.  This keeps additional stuff to an absolute minimum.  (P.S. - the Kivic runs on any voltage between 4.5 and 17v and comes with an adapter cable for direct wiring to the ship 12v power.) 

I have attached a few photos.  Feedback appreciated. 

 (I only have 2 hands and I needed to hold & move my phone and the Stratux at the same time... :)

I will be uploading the source for all of this to my fork at N120BZ/stratux by this coming Monday.
[image]
[image]
[image]
[image]