cyoung/stratux #768: Raspberry Pi 3 B+ testing

cyoung, 2019-02-25:
Testing image: https://www.reddit.com/r/stratux/comments/85nqif/rpi3b_image/

After writing the image, the main .sh updates can be used to bring the image up to v1.5b2. The .sh updates should fix the i2c issues, as well.

Anyone that has done testing, please add your experiences here.

jocamero, 2019-03-08:
I’ve been using a 3 B+ for many months now without any (related?) issues.  Any specific feedback you’re looking for?

K2AAE, 2019-03-14:
I'm still trying to track down documentation; The update process, since the image is 1.4.4 there is 1.4.5, 1.5.1 and now 1.5.2 as of 3/13. I uploaded 1.5.2 and nothing happened. I downloaded then uploaded 1.4.5 still nothing. With the release image 1.5.1 I downloaded 1.5.2 and got nothing then I sshftp 1.5.2 to /root chmod to 755 and rebooted that resulted in an update. Is there a location where Stratux is documented?

K2AAE, 2019-03-14:
I just confirmed the process: sshftp from desktop to /home/pi v1.4.5 then login as root on the stratux, cp the update to /root and chmod 775 the .sh shut down and when you restart after awhile the web browser can connect to the updated stratux. Only 2 more times to get up to v1.5.2
What happened to the OS reboot? It hits shutdown and stops, no reboot anymore. Every boot I get /usr/bin/stratux-screen.py: not found

jocamero, 2019-03-14:
Per the Wiki here's the link to the update process:  https://imgur.com/a/sKsjt
I don't seem to recall it being an incremental update process.  I went directly to the latest version.  Also, I think the only supported way to update is via the web GUI and the .sh files unless something has changed.

tl;dr - get the latest .sh file here:  https://github.com/cyoung/stratux/releases/ and upload via the web GUI

> I'm still trying to track down documentation

K2AAE, 2019-03-14:
OMG! The same 'button' that uploads the file has to be clicked to complete the process.... 30 years of compute'n and I can't remember seeing that used in a UI, ever. I just found out this morning that the WiFi is broken, seems the method I used to update killed it, I now have 192.168.10.12 but can't connect to the webpage. Etcher time again. 
This is really a great product except for the How To that was left out, having information spread all around the internet does not make a How To. Github, imgur, slack, reddit, stratux.me and numerous web sites.

apr911, 2019-03-15:
Broke my Pi3B and wasn't paying attention/didn't realize it wasn't fully supported when I bought the new one which was a 3B+. I have a spare 3B at home but I'm currently on the road so for now it's make do with the 3B+ until I get home this time next month.

It seems to take forever to connect to any towers and when I do connect, I usually only get a "marginal" connection at best.

I am using the vk-162 external GPS (works great so at least I still get track logs) and the v2 antennas (want to get a 978v3 but doesn't fit the summitlink case and would need to wait until I have a normal 3B or have resolved this issue first to limit variables anyway) 

There are a couple variables I am trying to work through but thought I'd post the comment here and see if there were any suggestions.

Variables to work through:
1) Confirm Antennas are correct. - I didn't really change my build aside from swapping out the board but it's possible the 1080 and 978 SDR's got swapped around. That being said, I can just barely make out the markings on the 978 SDR so Im fairly sure I have the antenna's paired to the correct SDR and I do pick up equipped traffic, just cant seem to get towers for the non-equipped traffic and weather.

2) Try new Location/Altitudes. As noted by 3b broke on a trip and I'm unfamiliar with reception in my current location (KISM/KGIF). After looking at the ADSB tower maps, I appear to be closer to some towers but further from others than I was back home at KMQS/KOQN/KPTW in PA and 47N/N51/KSMQ in NJ. I know the tower types and altitudes vary so I may just not be close enough to a low-altitude tower in my new location. My most recent flights have also not gotten particularly high usually staying in the 2000-3000 ft range but there is no "terrain" to speak of in the area and I usually get the marginal signal coverage at altitudes of 1000 ft when I am coming back closer in to KISM while get no signal on my departure from KISM and while flying between 2000 and 3000 ft in the vicinity of Davenport, FL and Winter Haven (KGIF) airport. In PA and NJ I've gotten solid connections to multiple towers once a few hundred feet in the air.

3) Try base 1.4r4 build and move up incrementally. I did an incremental upgrade from the "release" version 1.4r4 to 1.4r5 and then on to 1.5r1 and 1.5r2 but didnt stop to do any incremental version testing. As a result it is unclear if the issue existed with the base build or started after one of the upgrades.

4) Other ideas?

I'm reimaging the 1.4r4 base build and will give that a try on my flight this afternoon. I'll lose AHRS but considering I dont have a mount for the stratux in my airplane, the AHRS data I get isn't the most accurate anyway and it will help to at least eliminate a variable in the form of version and/or update process.

Cross-posted to Reddit Conversation.

wa9als, 2019-03-15:
Pi 3B works great for me and 1.5b2. The new radios work great (after I exchanged once with Amazon). It appears I can confirm what others have said about new uat radio not only being lower power, but has increased sensitivity for towers and keeping them. Yay!

K2AAE, 2019-03-17:
Installed, expanded SD which no longer happens during boot, did apt update and upgrade had errors on wpa not finding a config file and exiting. Following the updated wiki installed go, edited the .bashrc and added CFLAGS stuff that isn't listed in the wiki. Tried git but git was not installed, apt install git could not find git, did update and upgrade a second time then was able in install git. put stratux git /root followed the wiki but make produced:
root@Stratux:~/stratux# make
make xdump978 xdump1090 xgen_gdl90 fancontrol
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
Package librtlsdr was not found in the pkg-config search path.
Perhaps you should add the directory containing `librtlsdr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'librtlsdr' found
dump1090.c:52:21: fatal error: rtl-sdr.h: No such file or directory
 #include <rtl-sdr.h>
                     ^
compilation terminated.
Makefile:37: recipe for target 'dump1090.o' failed
make[2]: *** [dump1090.o] Error 1
make[2]: Leaving directory '/root/stratux/dump1090'
Makefile:25: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 2
make[1]: Leaving directory '/root/stratux'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
root@Stratux:~/stratux#

For crying out load, could you PLEASE list the dependent sources, the git locations and where they have to be? Any environmental variables that are needed like the paths and CFLAGS would be nice. I thought since this was a functional Stratux image that stuff (wiring, rtl-sdr) was already installed but guess not.

apr911, 2019-03-23:
Tried swapping the antennas and using the base build 1.4r4 still marginal.

Still could be location/altitude and/or interference. I'll be flying a short XC on Tuesday so we'll see if it improves on that flight...  If that doesn't do anything then I'm stuck until I am back home where I know I have good reception and can switch back to the regular 3B. I suppose it's also possible I damaged the radio's but wont know that until I can eliminate the other variables of location, 3B+ and image version.

gmint, 2019-06-05:
I used a B3+ for several months and had quite a few problems with the stability of the unit. Unfortunately, I *THINK* this was probably power related. After replacing my power adapter, cable, and ultimately switching back to a 3B, I once again have a stable connection. Of course, since the 3B+ takes more power, that might have been exacerbated by that fact. Otherwise, everything seemed to work as expected.

cyoung, 2019-06-05:
@gmint - Were you using the above testing image? Which radios and accessories were you using? If using two NooElecs, the unit was probably drawing over 1A total. That brings us back to our early power issues.

gmint, 2019-06-05:
@cyoung - I was using the testing image referenced which I brought up to 1.5b2 with the .sh updates. I was using the v2 radios sold by stratux on Amazon (not the NooElecs). Honestly, I really think it was the 12v power supply I was using (my plane is old enough that it has a cigarette lighter) but when it started giving me issues, I started replacing cables, and using different adapters and all of that stuff. Eventually I just decided to go back to the 3B to really eliminate all of the possibilities. After that, I ended up swapping the power adapter after that and I've had a stable connection so far (I'm using ethernet instead of WiFi). Unfortunately I don't have great data for isolating any issues strictly to the 3B+. At the worst, I think the 3B+ just made it worse since I did seem to have good luck for a period of time.

Personally, I wish they'd ditch the micro-USB for power. Stratux is not the first time I've had power issues on a RPi and I think a ton of that is due to the tiny connector. Maybe we'll get USB-C on the RPi one day. Also something that crossed my mind—have you considered the possibility of Stratux on something more "industrial grade" like the balenaFin device?

cyoung, 2019-06-05:
>Stratux is not the first time I've had power issues on a RPi and I think a ton of that is due to the tiny connector. 

Most of our power issues are from people using junk cables. The connector is fine to 2A or more.

gmint, 2019-06-05:
> > Stratux is not the first time I've had power issues on a RPi and I think a ton of that is due to the tiny connector.
> 
> Most of our power issues are from people using junk cables. The connector is fine to 2A or more.

Oh I agree that technically the connector is fine, but in real world use, over time it becomes loose and very prone to becoming dislodged.

rhole, 2019-06-05:
I would suggest that in aircraft use with the relatively high vibration environment that no push-in connector be installed without providing support for the inserted connector.  The installer should provide both strain and vibration relief.  This applies to, including but not limited, micro-USB, USB, the small polar power connectors.  Anything which can be pulled out will, eventually, fall out.

 

 

Rick Hole

gmint, 2019-06-06:
> I would suggest that in aircraft use with the relatively high vibration environment that no push-in connector be installed without providing support for the inserted connector. The installer should [...]

I agree fully. I will likely hot glue the connector in once I'm done tinkering around with my unit.

pvanessen, 2019-06-28:
won't compile on my 3B+ :( 

> make xdump978 xdump1090 xgen_gdl90 fancontrol
> make[1]: Entering directory '/home/phillip/stratux'
> cd dump978 && make lib
> make[2]: Entering directory '/home/phillip/stratux/dump978'
> gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
> gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
> make[2]: Leaving directory '/home/phillip/stratux/dump978'
> sudo cp -f ./libdump978.so /usr/lib/libdump978.so
> sudo: unable to resolve host stratux
> git submodule update --init
> Submodule 'dump1090' (https://github.com/stratux/dump1090) registered for path 'dump1090'
> Submodule 'goflying' (https://github.com/cyoung/goflying) registered for path 'goflying'
> Cloning into '/home/phillip/stratux/dump1090'...
> Cloning into '/home/phillip/stratux/goflying'...
> Submodule path 'dump1090': checked out '9a4fb850937565cfeadd1e5889cddbf93f45faf5'
> Submodule path 'goflying': checked out '0ba3e51be74b4848488e06ec0fee74ebaa7ef705'
> cd dump1090 && make
> make[2]: Entering directory '/home/phillip/stratux/dump1090'
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c anet.c -o anet.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c interactive.c -o interactive.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_ac.c -o mode_ac.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c mode_s.c -o mode_s.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c net_io.c -o net_io.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c crc.c -o crc.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2000.c -o demod_2000.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c demod_2400.c -o demod_2400.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c stats.c -o stats.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c cpr.c -o cpr.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c icao_filter.c -o icao_filter.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c track.c -o track.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c util.c -o util.o
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c convert.c -o convert.o
> gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o demod_2000.o demod_2400.o stats.o cpr.o icao_filter.o track.o util.o convert.o -lpthread -lm -lrt `pkg-config --libs librtlsdr` 
> fatal: No names found, cannot describe anything.
> gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c view1090.c -o view1090.o
> gcc -g -o view1090 view1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o stats.o cpr.o icao_filter.o track.o util.o -lpthread -lm -lrt 
> make[2]: Leaving directory '/home/phillip/stratux/dump1090'
> go get -t -d -v ./main ./godump978 ./uatparse ./sensors
> github.com/skelterjohn/go.matrix (download)
> github.com/gorilla/websocket (download)
> github.com/kidoman/embd (download)
> github.com/golang/glog (download)
> github.com/dustin/go-humanize (download)
> github.com/jpoirier/gortlsdr (download)
> github.com/mattn/go-sqlite3 (download)
> github.com/prometheus/client_golang (download)
> github.com/beorn7/perks (download)
> github.com/golang/protobuf (download)
> github.com/prometheus/client_model (download)
> github.com/prometheus/common (download)
> github.com/matttproud/golang_protobuf_extensions (download)
> github.com/prometheus/procfs (download)
[... 17 lines cut ...]
> Parsing meta tags from https://golang.org/x/net?go-get=1 (status code 200)
> golang.org/x/net (download)
> Fetching https://golang.org/x/net/ipv4?go-get=1
> Parsing meta tags from https://golang.org/x/net/ipv4?go-get=1 (status code 200)
> get "golang.org/x/net/ipv4": found meta tag get.metaImport{Prefix:"golang.org/x/net", VCS:"git", RepoRoot:"https://go.googlesource.com/net"} at https://golang.org/x/net/ipv4?go-get=1
> get "golang.org/x/net/ipv4": verifying non-authoritative meta tag
> Fetching https://golang.org/x/net/websocket?go-get=1
> Parsing meta tags from https://golang.org/x/net/websocket?go-get=1 (status code 200)
> get "golang.org/x/net/websocket": found meta tag get.metaImport{Prefix:"golang.org/x/net", VCS:"git", RepoRoot:"https://go.googlesource.com/net"} at https://golang.org/x/net/websocket?go-get=1
> get "golang.org/x/net/websocket": verifying non-authoritative meta tag
[... cut ...]
> /tmp/go-link-547846821/000002.o: In function `_cgo_b0c710f30cfd_C2func_getaddrinfo':
> cgo_unix.cgo2.c:(.text+0x678): warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
> collect2: error: ld returned 1 exit status
> 
> Makefile:21: recipe for target 'fancontrol' failed
> make[1]: *** [fancontrol] Error 2
> make[1]: Leaving directory '/home/phillip/stratux'
> Makefile:14: recipe for target 'all' failed
> make: *** [all] Error 2

Ergonomicmike, 2019-09-07:
I ran a B+ for months. Never had a problem (that I know of). Like @jocamero , am wondering what problems one should be on the alert for.

Ergonomicmike, 2019-09-23:
I noticed that Fry's Electronics is selling the Pi 3 B+ for $13, on sale clearance. 

That model didn't last long.

What's wrong with the B+?

wa9als, 2019-09-23:
Fry’s heard it doesn’t work with Stratux.

ssokol, 2019-09-23:
Biggest complaint with the 3B+ is the power draw. It takes quite a bit more than the 3B, and you don't get a really significant bump in performance.

That said, I think it's more an issue for Frye's than it is for the 3B+. They want to clear it out to make way for the Pi 4. (Not that they're short of available shelf space these days....) From what I recall, the RPi Foundation has committed to continuing the 3B+ for several years.

JohnMarzulli, 2019-09-23:
The Pi3B+ makes an excellent processor for my HUD project if you don’t want to run the HUD on the Stratux itself.

Ergonomicmike, 2019-09-23:
Okay on the power draw. I never noticed it. (Perhaps offset by the lower power SDR's.) I liked the B+ because it was quiet, RFI-wise, in the Glasair. I didn't do an exhaustive check of freqs, but I never had any noticeable Stratux-noise in the NAV/COM with the B+. 

The Fry's store near me (Phx) is HUGE. And empty of people most of the time nowadays. Also empty of stock. I'm guessing it must cost $50,000/mo just in electricity to air condition the building in our summers. I can't image how the store can stay in business much longer.

ssokol, 2019-09-23:
The latest batch of B3 units I've purchased have had the same metal shield caps on the primary ICs that the B3+ has. I've not done a side-by-side check, but I expect that will help bring the EMI level down.

Yeah, Fry's here (Bay Area) are all ghost towns. A reversal of fortune on almost the same level as Blockbuster Video...

Ergonomicmike, 2019-09-30:
Here's another problem with the B+ that's not software related, but makes for a reliability problem in airplanes. The metal shield that surrounds the microSD card can detach from the PCB. The shield acts as part of the capture mechanism/cage for the card itself. So if the cage breaks, no electrical contact for the card. (Found this out after our Glasair purchaser sent me my Stratux back.)

apr911, 2019-09-30:
@Ergonomicmike the hardware problem you describe exists in at least the 3b, 3b+ and even the new 4.

Its how I ended up using a 3b+ on my trip in the first place; the sd card cage on the 3b I had with me separated from the pcb and prevented the card and software from being read. Either I messed up ordering or the company I ordered a replacement from shipped the wrong one because I received a 3b+.

Ergonomicmike, 2019-10-08:
Thanks. I'm confused because, IIRC, the 3B has the spring loaded "clicker" to hold the card in place. Whereas my card is held in by friction on the B+. 

In any event, I'm going to try to solder the tangs down today on the B+. (If I can find high power magnifiers.)

Ergonomicmike, 2019-10-10:
Soldering the tangs back down to the PCB worked. My 3 B+ Stratux is back up, running 1.6r1. (And I hot glued the micro-antenna connector in place on the 978 SDR.)

Marliwahoo, 2020-03-21:
I'd love to try a version of stratux hat would work with the rpi4. Instructions or a working image would be great.

VirusPilot, 2020-03-21:
A while ago the following worked for Pi4:
- sudo -s
- rm /usr/local/lib/librtlsdr.*
- rm /usr/local/include/rtl-sdr*
- apt install librtlsdr0 librtlsdr-dev
- cd /tmp
- wget https://project- downloads.drogon.net/wiringpi-latest.deb
- sudo dpkg -i wiringpi-latest.deb
- cd ~/stratux
- make -B && make install

RFC!!

MysticCobra, 2020-04-18:
> Thanks. I'm confused because, IIRC, the 3B has the spring loaded "clicker" to hold the card in place. Whereas my card is held in by friction on the B+.

The 3B used a friction-fit SD card slot.  Spring loaded slots were last used in the RPi 2 series.

nikita-visnevski, 2020-05-26:
Folks, I am having a problem getting correct RPi 3B at this point.  I order it and get 3B+ in mail instead.  I returned two to Amazon already and have given up on them.  Ordered directly from CanaKit.  Explicitly picked Pi 3B as they had 3B+ as a separate option.  After 14 days of waiting, received it today, opened, and it is 3B+.  My point is, can no longer get 3B reliably.  I understand all the power issues with 3B+ and 4, but this is the reality going forward.  Soon these will be the only boards available.  I personally power it off the cigarette plug in the plane, so depleting the battery is no real concern.  Any way we can get stable release supporting 3B+?  I tried to mess with this experimental build.  Something is not right with it.  Just voicing frustration over the situation.  Are there any fundamental issues preventing Stratux support of 3B+ and 4?  If not, any way to actually reliably supporting it?  Thanks

cyoung, 2020-05-26:
@nikita-visnevski - What issues did you have with the experimental build?

nikita-visnevski, 2020-05-26:
AHRS does not seem to work for me in that one.  I think it is based off of the earlier version of Stratux which did not support full AHRS protocol compatible with ForeFlight.  It could be I am doing something wrong too.

eagsc7, 2021-11-02:
Does anyone know the Status of updates to the Stratux software for putting into the Pi 4?  Will cross-post into GitHub and Discord...

Is this close to a "v.1 Release"

VirusPilot, 2021-11-02:
Please consider this fork: [https://github.com/b3nn0/stratux](url). Many things have been implemented, including Pi4 support. The main motivation was to adapt for EU purposed like FLARM but in the meantime it is also the basis for the latest US images as well: [https://github.com/cyoung/stratux/releases/tag/v1.6r1-eu026-us](url)