cyoung/stratux #769: some general questions...

duecedriver, 2019-02-27:
1. Stratux version:latest
2. Stratux config:

    SDR
      - [ x] single
      - [ ] dual

    GPS
      - [ ] yes
      - [x ] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro9

    EFB platform: ( iOS )

    EFB hardware: ( iPad air)

4. Description of your issue:

I have been out of this project of a couple years now.. was previously running 1.0x with the original blue nesdr mini 2.

I did the .sh update to the latest 1.5b2 and have a few questions

after a couple page refreshes on the stratux status and settings pages.. I have lost the radio buttons to change from 1090 to 987 

the settings panel now only has AHRS, Commands, Configuration (modes, watch, ppm, ips) and wifi settings.. no radio controls at all

the box is essentially locked in 1090 mode and its pulling OK traffic.. probably not everything in the area based off flight aware maps.. it could be pulling more.

I am guessing that in order to keep using this project ... 

I assume the pi 2 B is OK, not great, but will work?

I assume that I should update to the latest v2/3 radio combo with antennas from amazon?

or did something not update fully/correctly?

Thanks

cyoung, 2019-02-27:
Hey @duecedriver - I remember you. Welcome back. You don't need to upgrade to the newer v3 radio, but its power requirements are much lower than the SDRs and it's quite a bit more sensitive. It performs better at higher altitudes. The Pi2 still should work just fine as well. Upgrading will just get you more battery life and better reception.

About the radio buttons, you'll have to enable "Developer Mode". On the main page, tap the version number rapidly about 10 times. There is no way to get out of "Developer Mode", but all it does is enable more options on the Settings menu.

duecedriver, 2019-02-27:
@cyoung 

thanks....

ok.. that works... the green light blinks now and is not steady so I assume that is related to developer mode?

performance with this old sdr seems pretty performant.. so some nice software improvements helping there

in this beta if show source in callsign is selected.. no callsign is shown in wingx v9.  turning that off and callsigns show up again in wingx

i see that on amazon there is a package of 2 a v2 and v3 with antennas.. I assume that is the best way to go moving forward...

Helno, 2019-02-27:
> i see that on amazon there is a package of 2 a v2 and v3 with antennas.. I assume that is the best way to go moving forward...

The V3 UATradio works way better at altitude than SDR's and drops the power usage down quite a bit.