cyoung/stratux #770: Update https://github.com/cyoung/stratux/wiki/How-to-compile-and-build-Stratux-executable

jackgavin, 2019-02-27:
1. Stratux version: Current
2. Stratux config: N/A

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: USB Internal GPS

    AHRS
      - [X] yes
      - [ ] no

    power source: EasyACC Battery

    usb cable: EaseACC Cable

3. EFB app and version: Foreflight Mobile current version

    EFB platform: iOS 12.1.4

    EFB hardware: iPad Mini 2

4. Description of your issue: When somebody gets a chance, please update https://github.com/cyoung/stratux/wiki/How-to-compile-and-build-Stratux-executable. It doesn't work and the last contribution was in September 2018, which leads me to think that the compilers need to be updated.

MolsonB, 2019-03-14:
What isn't working, what errors do you get during 'make'?

The notes on the link look pretty close to mine. I've used a different GOVersion though, that's about it.

`wget https://dl.google.com/go/go1.9.linux-armv6l.tar.gz
tar -zxvf go1.9.linux-armv6l.tar.gz`

cyoung, 2019-03-14:
A compiler update isn't required. Jack was trying to compile using a clean Raspbian image and running into a lot of dependencies issues that weren't covered by the wiki.

I've added a note to the Wiki to clarify that the instructions are for users that start with a Stratux image. To keep our efforts from being spread too thin, we never really spent much effort in trying to make the software work with many different configurations. I don't see that changing.

Added: 

```The standard Stratux image has all of the dependencies installed except for those installed below. If you are using a clean Raspbian install or another distribution, you will need to dig through the `image/` directory in the source and manually determine what your system needs.
```

closed 2019-03-14 by cyoung