cyoung/stratux #771: Logging and Shutdown issues on v1.5b2

K2AAE, 2019-03-23:
1. Stratux version: v1.5v2
2. Stratux config: RPI 3

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:
M8T
    AHRS
      - [X ] yes
      - [ ] no

    power source:
3amp
    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
NC
    EFB platform: (e.g., iOS 9.2)
NC
    EFB hardware: (e.g., iPad Mini 2)
FireFox current from PC/Galaxy Note 10  
4. Description of your issue: 1. Since update to v5.2 no means to enable logging. 2. Shutdown hangs after Stopped Target Network for a while. For some reason the fan is hanging everything for 90 seconds. Can a button be attached to the AHRS to shut the unit off? GPIO4,17,18 ?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

K2AAE, 2019-03-23:
http://192.168.10.1/logs/stratux/ returns 404 page not found

Helno, 2019-03-26:
The logging options were moved to a hidden tab (The developer tab) since many people were enabling logging and filling/corrupting SD cards unintentionally.

If you wish to access the developer settings just go to the web interface and click on the version number a bunch of times and it will appear. 

I you want to add a button to shut it down via GPIO go right ahead and add it. You will need to look up how to call a script via a button press.