cyoung/stratux #772: Should serial connected GPS modules be supported?

Helno, 2019-04-13:
1. Stratux version: 1.5b2
2. Stratux config: Pi 3b

    SDR
      - [x] single
      - [ ] dual

    GPS
      - [ ] yes
      - [x] no
      type:

    AHRS
      - [ ] yes
      - [x] no

4. Description of your issue: With a fresh setup Stratux.log appears to be filling up as it attempts to configure the GPS. 

With no GPS connected it is running as though it has a serial connected GPS and fails repeatedly to apply the GPS configuration.

We have a wide variety of USB GPS modules supported and I don't recall anyone using serial GPS modules lately. Just seems like it causes unnecessary writes to the log file and the odds of it being useful to anyone is limited.

[stratux.log](https://github.com/cyoung/stratux/files/3075668/stratux.log)

JohnOCFII, 2019-04-13:
I still have one working Stratux using the GPIO connected GPS RY835AI.  But it is running old code, and maybe will just stay that way.

Helno, 2019-04-13:
I don't think there are many out there that wouldn't get by just fine with an older version of the software.

cyoung, 2019-06-05:
I've moved this log write to DEBUG only:

```UBX8 device detected on USB, or GPS serial connection in use. Attempting GLONASS and Galelio configuration.```

Related: #482.

saabnut, 2019-06-18:
Personally I would prefer it if Stratux simply harvested "ownship" broadcast from my ADS-B out transmission on 978 mhz. The data is FAA Certified WAAS with an external Certified antenna. This would eliminate ownship ghosting, and when the no GPS flag popped up, I would know that my 978mhz data was not current.

Helno, 2019-06-19:
> Personally I would prefer it if Stratux simply harvested "ownship" broadcast from my ADS-B out transmission on 978 mhz. The data is FAA Certified WAAS with an external Certified antenna. This would [...]

That sounds like a great idea but comes with a bunch of limitations. Not everyone has ADS-B OUT. Your ADS-B OUT might be fancy and certified but it will not have the same refresh rate as a GPS connected directly to Stratux.

It also wont fix this particular issue as it is just a case of Stratux thinking something is wrong and generating needless log file entries.

saabnut, 2019-06-19:
My ADS-B out is not fancy, just certified. (GDL-82) This would eliminate 
or at least backup the UBlox device, which is for sure not SIL3.

The refresh rate at ~ once per second is not that bad. The data 
presented would match the data presented to other aircraft.

It would give me an easy way to see if there is an discrepancy between 
the altimeter and encoder. That alone makes it worth doing.

Operating without ADS-B out for those who do not have it, comes with a 
bunch of limitations,

Helno, 2019-06-19:
>The refresh rate at ~ once per second is not that bad. 

The uBlox 8 we use refreshes at 10 hz. That is significant if you want to use it for AHRS. In this use case refresh rate is more important than an extremely precise position. 

Stratux already picks up your ADS-B out for ownship position when it detects it.

saabnut, 2019-06-19:
I understand completely, but this is about traffic for me. If stratus was
ready for rs 232 in, i would use my non waas c129a1 output to feed it.
Garmin/apollo
9600 baud nmea mapcom data is coming from that gps. 5hz update, built in
integrity warning RAIM. But again, the ads-b out unit is more accurate.
For ahrs, i would want pitot static data input and we are not there yet. I
would just as soon have a different stratux for that function, securely
mounted to the floor.

PepperJo, 2019-06-20:
> Operating without ADS-B out for those who do not have it, comes with a
> bunch of limitations,

Don't forget about people in Europe. ADS-B out is not mandatory here (and there is no deadline defined yet when it will be), so most GA airplanes only have Mode-S. Additionally, I rather have another independent system, e.g. when I loose power in my airplane.

saabnut, 2019-06-20:
I would never forget about Europeans, my ancestors. I have not been 
flying in small planes in Europe, but if I was operating there I would 
have a FLARM system, as well.

Without a ground station, how does a 1090 mode-s aircraft show up on 
stratux? There is no extended squitter data to show direction and 
velocity. Only altitude, squawk and ID.

We had passive systems that used transponder signal strength to 
calculate range, and antenna arrays to indentify relative position.

PepperJo, 2019-06-20:
Correct mode-S traffic does not show up. You can only get proximity warnings but I don't think they are very useful. uAvionix tried to implement the "passive system" in their SkyEcho but they deemed it to be too unreliable to be useful.

FLARM, yes, and we even have Stratux FLARM support however there are still a lot of people here flying without FLARM since it is not mandatory. Ontop of that FLARM has its own shortcomings that make it not a good fit for powered resp. faster airplanes (e.g. range).

gettyhub, 2021-10-17:
I have a gps that connects through a usb serial interface (all in one, not separate gps and serial), and it comes through /dev/ttyUSBx.  I just symlink it to /dev/prolific0 and it works just fine in NMEA mode.