cyoung/stratux #773: UBX-CFG-RATE Payload is incorrect for ublox 8

VirusPilot, 2019-04-18:
This the ublox 7 configuration:

updatespeed := []byte{0x64, 0x00, 0x01, 0x00, 0x01, 0x00} // 10Hz=100ms, cycle=1

and this is the ublox 8 configuration:

updatespeed = []byte{0x06, 0x00, 0xF4, 0x01, 0x01, 0x00} //

These bytes don't make any sense to me unless I missed something, 2Hz would mean 500ms -> 0xF4 0x01, followed by 0x01, 0x00 for cycle=1

corrected:

updatespeed := []byte{0xF4, 0x01, 0x01, 0x00, 0x01, 0x00} // 2Hz=500ms, cycle=1

https://github.com/cyoung/stratux/blob/50e2b3c9973ad004c9d88281cd811663c7a495ff/main/gps.go#L313

VirusPilot, 2019-04-25:
After quite some testing I suggest the following:

 // 100ms & 1 cycle -> 10Hz Navigation Update rate
p.Write(makeUBXCFG(0x06, 0x08, 6, []byte{0x64, 0x00, 0x01, 0x00, 0x01, 0x00}))

which works fine for both GSP/SBAS (e.g ublox 7) and GPS/GALILEO/GLONASS (ublox 8). In fact disabling SBAS for ublox 8 enhances accuracy.

PepperJo, 2019-04-25:
Why not add a button to the settings page to enable/disable SBAS? (We can only add it to developer mode if we think it is to much detail for normal users)

VirusPilot, 2023-08-14:
resolved since Stratux Europe Edition release

closed 2023-08-14 by VirusPilot