cyoung/stratux #774: Pitot static data

Qhergt, 2019-05-05:
I have an Arduino calculating airspeed, pressure alt with adjustable QNH and tilt compensated Compass. I'd like to link this to the stratux as to provide actual Pitot static and magnetic compass data to be broadcasted instead of the GPS data so AHARS in foreflight, naviator and other apps display actual airspeed, alt and mag heading. 

I have all the code done on the Arduino side, I'm not familiar with Python or programing Pi so I'd like to collaborate in order to have this as an option for Stratux with ability for user selection of data source via the stratux web or app interface. 

This device is very small and would be battery powered and able to attach via camera screw mount  could be easily attached to the outside or inside of an aircraft

DocTopi, 2019-05-12:
Integration these datas in Stratux and Stratux/AHRS woul be a very wise enhancement. Is there any possibility for this?

saabnut, 2019-06-18:
It would be great to have this functionality. Alternatively a pixhawk autopilot could provide this data.

Huge-Lead-Balls, 2019-10-23:
I'm looking to feed and display airpeed as well.   Anything new with this?

Qhergt, 2019-11-06:
Its in progress.

Qhergt, 2019-11-06:
Its in progress

javedrahman, 2019-11-14:
Excellent news. Is there any details on what hardware is required for the pitot-static data? Also, where is it mounted and how is it powered?

Qhergt, 2019-11-14:
Arduino based.  Battery powered, and most likely camera mount thread in box like the stratux.

Helno, 2019-12-16:
Any update on this? You were asking to talk to a developer on facebook but never reached out on here or Slack?

Qhergt, 2019-12-16:
Yes it's getting close.  Forgot to message.
Been a bit busy.

Qhergt, 2019-12-16:
There are a few ways I have an doing this.

#1 is that I can have an Arduino processing all the data and sending out MAG heading, Airspeed and pressure alt via Bluetooth to the stratux.

   Or

#2 I can go even smaller and have just the raw pitot pressure and static pressure values sent again by BT to the stratux but have stratux do the math to convert into Airspeed and pressure Alt.

Or

#3 Now, the stratux AHRS card already has an MPU9250 and BMP 280  sensors.  Both of which can be used to derrive magnetic heading (MPU)  and pressure alt (BMP, but in unpressurized aircraft)      There should be a way to access the MPU's 3 axis magnometer,gyro and accelerometer to get a tilt compensated magnetic compass heading.     Also the baro pressure from the BMP to get pressure alt.      This means only one sensor would be needed to gather pitot pressure for airspeed in non pressurized aircraft.

Depending which way you want to go with this will determine what Arduino id need to use, and the size of the overall package.

Right now for #2 it looks like the entire works could fit into a 1.5-2" wide by about 4" long enclosure with a 1/4"-20 unc thread like on the stratux case and stuck under an aircraft wing using a GoPro adhesive mount with the correct adapters to a 1/4"-20 stud.    This includes a 18650 battery shield to power the unit and also make it rechargeable.

Please let me know

Qhergt, 2019-12-16:
Oh, forgot to mention, I would be best if stratux did the actual pressure alt calculations because a user interface will be needed for altimeter setting.    Probably added to the stratux manager page.

saabnut, 2019-12-16:
Not to get ahead of your thought process, but have you seen the uavionix
av20s? Synthetic angle of attack...so to speak. Derived from pitch, bank,
airspeed, g force, time, etc.

Qhergt, 2019-12-16:
Yes, but have you seen the cost?
I built my own EFIS based on this
https://experimentalavionics.com/efis-19264-ver-1-2/.    For under $200 Canadian.
[Image]

Flown with it and it works great.
Altimeter bench tested at avionics shop and found to be well within certification limits.

This can be slimmed down quite easily
Consisting of Arduino nano, HC-06 Bluetooth module, MPXV70010dp and honeywell baro pressure sensor.   (Ill have a prototype done shortly)
And be built for around $100 Canadian. (The Honeywell sensor is $65 on its own)

After other things like custom PCB, enclosure, 18650 battery shield etc, under $150 for sure.

1-prop-head, 2019-12-17:
Would the esp8266 or esp32 reduce your chip count since they provide communication  via built in WiFi or Bluetooth (esp32)

Qhergt, 2019-12-17:
I guess I would.
However if stratux does all the math from the raw data, i could get it into a digispark ATTiny 85 that will plug directly into the 18650 sheild.

rv7charlie, 2019-12-21:
Hi; just found this thread while searching for ADAHRS on Stratux. Couple of thoughts from a non-programmer. I've been hoping someone would add pressure sensors (and code) to the stratux for static and pitot pressure. This could allow it to be a true ADAHRS (with air data) instead of simple AHRS; if an external temperature probe is included, we'd get everything a hi-dollar commercial system like big D & big G offer. 

But I feel I should point out that a baro sensor vented at the the Statux is likely to generate significant static errors, leading to both significant altitude errors and significant airspeed errors. Static port location on the airframe can be quite critical for accurate measurement. The RV-x a/c can see a 10kt error in airspeed, and 50-75' error in altitude, just from having a 'flush' static port vs it being on a 'bump' on the fuselage the thickness of a pop rivet head. With the static port in the cabin, any vent or heater vent being opened/closed will cause static to change, along with airspeed and altitude.

Thanks for your work; I wish that I could be of more help than just pointing out pitfalls....

Charlie

Qhergt, 2019-12-22:
Which is why Im.also suggesting an external one.    Although an off set can be added if using the internal one on the AHARS card.