cyoung/stratux #775: UDP/TCP ports: 4353, 10110, 4352, 2000, 23

dgofman, 2019-05-19:
Hi, I am using Stratux version:  v1.4r5 and would like to integrate with XCSoar 6.5.
By default XCSoar supporting UDP/TCP ports: 4353, 10110, 4352, 2000, 23.
I found old tickets:
https://github.com/cyoung/stratux/issues/489
https://github.com/cyoung/stratux/issues/571
https://cloud.githubusercontent.com/assets/8769091/23965570/a163ca32-0975-11e7-9efd-55d3a1ee31a1.png

Where you described supporting FLARM ports, however, I don't this that option anymore.
Please can you add this feature back,

cyoung, 2019-06-05:
We never added FLARM support to the main branch.

Try this fork: https://github.com/PepperJo/stratux.

closed 2019-06-05 by cyoung

VirusPilot, 2019-06-05:
In fact this fork here is the "European" Stratux flavor:

https://github.com/b3nn0/stratux

It includes all previous FLARM developments in one fork.

dgofman, 2019-06-09:
Yes, but my goal connecting startux and XCSoar. I am using your image and instruction from https://github.com/0sander/stratux_air_nav_pro  and cannot get Data. I tried to change a port to 4352 and ip address still no lack.
Please, can you help me?

dgofman, 2019-06-09:
[image]
[image]
[image]