cyoung/stratux #778: Stratux apple  app

jmiv66, 2019-06-11:
1. Stratux version:
2. Stratux config:1.5b2

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Anker Battery

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight

    EFB platform: (e.g., iOS 9.2) IOS 12.3.1

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 2

4. Description of your issue: Is the  Stratux Apple app still valid? App will not come up, only has white screen and can not see my statux. Connected to wifi and can be seen in Foreflight.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2019-06-11:
Are you talking about this app from OpenFlightSolutions?
https://www.openflightsolutions.com/flightbox/app/

That is not maintained by the developers on this GitHub project. You'll have to contact OpenFlight Solutions support for help with that app.

cyoung, 2019-06-11:
Try browsing manually to http://192.168.10.1/. You might need to delete the "app" (shortcut) on your device and remake it.

closed 2019-12-16 by Helno