cyoung/stratux #779: Ghosting

topnotchaero, 2019-06-17:
1. Stratux version:1.5b2
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ ] yes
      - [X ] no
      type:

    AHRS
      - [X ] yes
      - [ ] no

    power source:

    usb cable:hard wired

3. EFB app and version: (e.g., WingX Pro7 8.6.2) FltPlan Go 4.7.22

    EFB platform: (e.g., iOS 9.2) iOS 12

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 2

4. Description of your issue: I have a Garmin GDL-82 out, and I'm still seeing a ghost of my aircraft.  Is there a way to rectify this?  I read that once the ADS-B out was installed, the issue would be resolved.  Thank you!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2019-06-19:
If your ownship isn't being picked up you can force it by entering your aircrafts hex code into the Stratux web interface.

Just look up your hex code on the aircraft register and enter it into the settings tab of the web interface.

topnotchaero, 2019-06-20:
Thank you!  

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

800.395.8386

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

topnotchaero, 2019-06-20:
Where would I find the option within the Stratux web interface?

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

800.395.8386

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

Helno, 2019-06-20:
Settings tab. 

Mode S hex code.

topnotchaero, 2019-06-20:
I don’t have a mode S.  Will entering the hex code in that field keep my GDL-82 from ghosting?

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

800.395.8386

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

Helno, 2019-06-20:
All aircraft have a mode S code assigned to them even if they do not have the hardware to broadcast it.

The shop that installed your GDL82 would have entered your hex code into it. This is what is broadcast along with your position to have ADS-B OUT. The only time this is not broadcast is when you use UAT anonymous mode.

topnotchaero, 2019-06-20:
I know what the hex code is.  I just wasn’t sure entering it into the Mode S field would solve my ghosting issue.  I’ll be at the airport this weekend and give it a try.  Thank you!

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

800.395.8386

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

topnotchaero, 2019-06-28:
It worked.  Thank you!

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

800.395.8386

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

closed 2019-07-02 by Helno