cyoung/stratux #780: Raspberry Pi 4

davidobrien1985, 2019-06-27:
Hi,

I'm wondering if it's already known if stratux will work with the Raspberry Pi 4.
I'm planning to use it with my AvPlan EFB here in Australia.

Also, just to get my understanding right. I won't need mobile coverage to use the ADSB receiver through the GPS antenna, right?
Right now, my EFB seems to only show me ADSB traffic if I also have mobile data reception.

Thanks!

flyadive, 2019-06-30:
David:

You are CORRECT.  You will  NOT need mobile coverage.  When you say mobile, you are referring to cell phone type service, correct?  NO Cell needed.   
And...  The GPS antenna is NOT required or used for RPi ADS-B In.  
The Traffic and Wx comes in on 978 Mhz 
The higher altitude Traffic comes in on 1090 Mhz.
Side Note:  A standard Transponder Antenna mounted on the belly of the plane works GREAT for the signals on 978 Mhz.  I used a length of RG-400 with BNC connectors and then converted it to SMA  connectors to mate with the SDR.  I am not concerned with High Altitude Commercial Traffic so I use the rubber ducky type antenna in the cabin...  Works good enough.

AND - I'm waiting for a knowledgeable person to respond to your RPi 4 question.  

Barry

heelix, 2019-06-30:
All these days... ty.  For the bug smashers I fly, the 978 probably covers
everything I'm trying to do.

wa9als, 2019-06-30:
There are a lot of low and slow bug smashers on 1090 out (only), so you definitely want dual band ADS-B In capability. The Stratux antennas are fine.

Helno, 2019-07-02:
The pi 4 is not likely to be officially supported because like the 3b+ it requires an OS upgrade and uses more power while providing no additional benefits for this use case. 

If anyone would like to see if the pi 4 will work with Stratux. You can always try upgrading the OS on a pi 3 and see if Stratux will still run afterwards. 

https://gist.github.com/nickfox-taterli/fcddd0ac724ec066b51f8e777800fea2

JohnMarzulli, 2019-07-02:
I have two Pi4s (2GB & 4GB) on the way to me with the intention of running the StratuxHud and Stratux on the same unit.

b3nn0, 2019-07-02:
If you want to give the Pi 4 a quick trial, you can also use the Stratux European Edition Fork Beta image, that is based on the Raspbian Buster release. It contains the same features as original Stratux, but some more Europe related Settings: https://github.com/b3nn0/stratux/releases/tag/v1.5b2-eu006
While the image works fine on the b and b+, I have not tested it on the Pi4, and I think in its current form, it will not work:
https://lb.raspberrypi.org/forums/viewtopic.php?t=243982

fhteagle, 2019-07-12:
I would very much like to see the Raspi 4 supported. I think the ability to carry one raspi4 that can be a thin client / desktop, media center, and then also serve as the stratux unit while flying is a very appealing proposition to us heavy road warrior pilot types. Yes, this can be done with a raspi3, but with far less performance and more lag than a 4. Therefore, I think the "no more capability for far more power draw" argument is far outweighed by having one multipurpose device...

I will check out what @b3nn0 suggested for sure. If anyone else knows of a good route to a working stratux on raspi4 setup, please post here. Thanks!

closed 2019-07-21 by davidobrien1985

ninelima, 2019-12-27:
ADS-B in Australia does not support 978 and has no weather.
You can display a HUD directly with Stratux and [Kwik EFIS](http://members.iinet.net.au/~ninelima/efis/index.html)
[image]

gettyhub, 2021-04-27:
Stratux European Editon works with the RPi 4.