cyoung/stratux #781: STRATUX ADS-B In and ForeFlight

flyadive, 2019-06-30:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2019-07-02:
I think you forgot to actually let us know what the issue is?

flyadive, 2019-07-02:
Helno and All others that responded:

THANK YOU!
As of now, there are no issues. I was thinking forward.  As for the RPi 4
they did make quite a few changes, just look at all the new size
connections they installed.  More dang plugs and adaptors to deal with.

Thanks Gang,

Barry

closed 2019-07-09 by Helno