cyoung/stratux #782: Help. How to get ADS-B weather data from Stratux?

siddharthsure, 2019-07-02:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [-] dual

    GPS
      - [-] yes
      - [ ] no
      type:

    AHRS
      - [-] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
Hi, i want to have the raw live ADS-B weather data (In any format) from stratux to integrate with Matlab. Is it possible? If yes, Please let me know how to. I tried searching in reddit and there is no answer. Thank you.   
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2019-07-09:
Take a look at the APP vendor integration guide. 

https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md

185Tom, 2019-09-13:
You have to be within a mile or two from a tower to receive UAT ADS-B wx info on the ground

closed 2020-08-16 by Helno