cyoung/stratux #786: WingX and ARS issues

ScraboTower, 2019-07-28:
1. Stratux version:  1.5b2
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: Stratux GPYes 2.0 u-blox 8 GPS unit 

    AHRS
      - [X ] yes   - Stratux AHRS Sensors and Fan Controller/Raspberry Pi MPU9250+BMP280
      - [ ] no

    power source:  usb from Ships Power

    usb cable:  Anker

3. EFB app and version: (e.g., WingX Pro7 8.6.2)    WingX 9.0.1.1

    EFB platform: (e.g., iOS 9.2)    IOS 12.4

    EFB hardware: (e.g., iPad Mini 2)  AIR 2

4. Description of your issue:

Just swapped out the old RY835AI combo AHRS/GPS chip for the Stratux GPYes 2.0 u-blox 8 GPS unit  & Pi MPU9250+BMP280.   

GPS is fine, Stratux status page allows me to set AHRS orientation and AHRS is behaving correctly  within the AHRS status page - however in WingX both the SV and Attitude Indicator screens flash continuous yet WingX shows it is receiving AHRS data.   
[image]
[image]
[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

ghainbach, 2019-12-09:
e-mail sent to Hilton Software - 12/08/2019

Hi WingX 9.1.0.7 support,
FYI ... I have loaded Stratux v1.4r4 (a1ea17e7ca) and can confirm that AHRS is working correctly with WingX 9.1.0.7.

I will contact Stratux and let them know that ForeFlight support added in v1.4r5 has broken WingX AHRS support.

Regards Graham

Pilot135pd, 2020-01-27:
first of all thank you ghainbach because when I read your post I have WingX Synthetic Vision working again. So I tested a few other versions and it seems like v1.6r1 is also working with WingX version 9.1.2.1 which I just downloaded yesterday.  Have you tried it?

ninelima, 2020-01-28:
In addition to WingX you might have a look at [Kwik EFIS](http://members.iinet.net.au/~ninelima/efis/) as well. It is fully open source and supports Stratux.

Pilot135pd, 2020-01-28:
I'll look into that, thanks.

UPDATE regarding  v1.6r1 : It went nuts on me yesterday in the air so I'm back to using v1.4r4 again with WingX 9.1.2.1

Pilot135pd, 2020-01-28:
> In addition to WingX you might have a look at [Kwik EFIS](http://members.iinet.net.au/~ninelima/efis/) as well. It is fully open source and supports Stratux.

I'll look into that, thanks.

UPDATE regarding v1.6r1 : It went nuts on me yesterday in the air so I'm back to using v1.4r4 again with WingX 9.1.2.1

Pilot135pd, 2020-01-28:
> In addition to WingX you might have a look at [Kwik EFIS](http://members.iinet.net.au/~ninelima/efis/) as well. It is fully open source and supports Stratux.

What a shame, Kwik EFIS isn't for iPads, only Android.  Do you know any other on IOS that the Stratus works correctly with?