cyoung/stratux #788: Help. What is encoding for GDL90 messages over WebSocket

femski, 2019-08-07:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

   I am the developer of AvNav so its about AvNav

    EFB platform: (e.g., iOS 9.2) iOS and Android

    EFB hardware: (e.g., iPad Mini 2) Mini, iPAD etc

4. Description of your issue:

I am trying to decode the ws://192.168.10.1/gdl90 WebSocket messages from Javascript App in iOS safari (ours is a PWA). It seems Stratux does not set message type to Binary so they end up as "String" in iOS Safari client. After that its difficult to figure just how to decode this string - I have tried various ways including TextEncoder but data does not come out correct.

It would have been nice if message type was binary because browsers can then receive it as binary. I am not asking to change this at this point, but just want to know how to decode a byte[] sent via Go Websocket that ends up as string in javascript WebSocket client! To me it seems incompatible

Any ideas ?

femski, 2019-08-07:
My code is like this:

let client = new ReconnectingWebsocket("ws://192.168.10.1/gdl90"); 
client.binaryType = 'arraybuffer'
client.onmessage = (event: MessageEvent) => {
   console.log("ADSB  Buffer is String: " + event.type)
   console.log("ADSB data: " + event.data
})

Stratux send "strings" like this and below and encoder like

data = new TextEncoder("utf-8").encode(event.data)

or any other encoder/decoder does nt seem to have produced valid bytestream.

"fhQQoNN0HUWKy29GKKmJGJAWxANBQUwxNDg3AABG1X4="
"fgARAfqhAADiA34="
"fhQQoNN0HUWOy28RKKmJGKAWwwNBQUwxNDg3AAAIFH4="
"flNYAQEBBQEC/////wH4AAYAAwAAAAMAAAEiAZcAonh+"
"fhQQoNN0HUWOy28RKKmJGKAWwwNBQUwxNDg3AAAIFH4="

femski, 2019-08-14:
Figured its base64 encoded with some extra quote characters.
Closing.
Thanks.

closed 2019-08-14 by femski

stevecagle59, 2020-12-02:
Hi,  Did you resolve this GDL90 decoding problem?  If I do a base64 decode the data doesn't match the message description for a GDL90 heartbeat message lets say.
Thanks,
Steve