cyoung/stratux #789: Issue compiling with GOROOT and Make on C.H.I.P.

BrianAnim, 2019-08-12:
1. Stratux version: Not sure how to tell
2. Stratux config:

Unit: C.H.I.P. PC
Reddit threads: https://redd.it/c3ln0y
https://redd.it/co8tke

    Power source: 5v wall plug

3. EFB app and version: FltPlanGo

    EFB platform: Android 8

    EFB hardware: Samsung Tab a8 4g (verizon)

4. Description of your issue:

When I get to the portion within these steps:
https://github.com/cyoung/stratux/issues/787

> • update .bashrc with GOROOT, GOPATH, PATH, CGO_FLAGS_ALLOW as follows:
> export GOPATH=/home/pi/go
> export GOROOT=/usr/lib/go-1.11 (use whatever comes back from go env | grep GOROOT)
> export CGO_CFLAGS_ALLOW=-L/opt/stratux/stratux_src
> export PATH=$GOPATH/bin:$PATH
> • log out and back in or source the .bashrc file
> • edit the dump1090 make file and remove flag Werror as there is a warning for data truncation that prevents compiling... "interactive.c:140:44: error: '%03d' directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]"

>     cd /opt/stratux/stratux_src make • remove check from Makefile for GOROOT as it isn't detecting a set env var during make install sudo make install

And i get: "goroot is not set! stop!"

If I comment out goroot in the makefile it runs, but when I try sudo make install I get Recipe for target 'all' failed make: *** [all] Error 2

cyoung, 2019-08-12:
Here were some scripts made by people that got Stratux running on the C.H.I.P. That was a while ago.

https://github.com/jpoirier/stratux-setup

I guess you're aware that the C.H.I.P. is a dead platform, so I have to ask. Why go through the trouble? Besides being out of production, the hardware and software never reached a stable state.

BrianAnim, 2019-08-12:
Because I feel like it’s the only thing I could use this for that would actually make it worth purchasing. Trying to save money and get something out of this.

https://github.com/jpoirier/stratux-setup

cyoung, 2019-08-12:
I think you might just write off the $9. It's not worth the time investment. I've had one C.H.I.P. last about 2 hours of use before having an unknown hardware failure. Not an uncommon occurrence if you search around.

BrianAnim, 2019-08-12:
I want to say I tried this version of stratux-setup.sh and that's how at least it's partially installed, with wifi out under the stratux SSID but no web page to browse to on 192.168.10.1.
Do you think I also need to run chip-power.sh and chip.sh ?

cyoung, 2019-08-12:
Probably. Closing the issue since this is a dead end project that likely won't benefit anyone in the future. The original issue about GOROTO not being set is not even a Stratux-specific issue. Search Google for "goroot is not set". Look in the image/ directory of the source here, maybe use the `.bashrc` that is included and review `mkimg.sh`. Good luck!

closed 2019-08-12 by cyoung

Thore-Krug, 2019-08-27:
Even though the C.H.I.P Platform doesnt have official support doesnt mean that the community isnt working on it.