cyoung/stratux #790: Strayux TLS/SSL/HTTPS support

femski, 2019-08-14:
I am the developer of AvNav - we are soon launching an EFB in PWA (Progressive Web App) format - and Stratux will be the only ADSB supported initially - because it has WebSockets support. App is working now - it works same on iOS an Android and all the desktops and works in cockpit without internet.

To make ADSB happen though: we need HTTPS/SSL enabled in Stratux. Without this Stratux output is considered "mixed content"/insecure and blocked by Safari/Chrome etc. You can not have a PWA coming from a secure domain (https://xxx.com) access ws://192.168.10.1/gdl90.

This means Stratux Go server will need to register a certificate on start. I am researching SSL for IoT options - but it seems a custom build in short run to incorporate SSL - but long term I think Stratux should include its own SSL/TLS support out of box.

This is a simple change to ListenAndServe call with extra arguments for cert files. Stratux can then be access vis https and wss://adsb.starus.me/... or whatever Stratux choses to register.

How should we go about it? I am getting setup to change the Stratux code, but we should perhaps discuss - how the community wants to prepare for ongoing revolution of PWAs (no more slow, expensive development for iOS/Android) . SSL/HTTPS issue is the center fo this. What subdomain should be used? Who will register it? How to mange validity of cert for life of device? etc. Thanks.
if this outside scope o Github issue then we can discuss here:

https://www.reddit.com/r/stratux/comments/cq89gl/stratux_adsb_and_coming_generation_of/

femski, 2019-08-15:
Update:

I registered a public sub-domain and got an SSL certificate issued to it by Go-daddy - mapped sub-domain to 192.168.10.1 IP address and App connected to Stratux via wss fine.

There are several issues with this approach: use of custom sub-domain, expiring certificate and mapping of a public sub-domain to a private IP (a questionable tactics at best).

So next thing I am trying is to use a self-signed certificate and install it on iOS and have iOS trust it as a root certificate. I expect it to work and if so that would be probably be the way to go:
a) no certificate expiration issue b) use a generic easily recognizable CName like "stratux.local" - or whatever Stratux choses to use

So far I have had to change less than 5 lines of code: basically serve both http and https content at standard ports (80 and 443) simultaneously. Use wss://... in place of ws:// (from the app only) and it all works. http:// site all works fine and there is  to impact one existing web/http/ws interfaces.

@cyoung 
Should I/can I submit a pull request when done ?

This seems like a very simple thing and it will open up Stratux to whole world of PWAs - it does not disturb existing http interface at all.

cyoung, 2019-08-16:
@femski:

>mapping of a public sub-domain to a private IP

Curious about how you plan to have the PWA set up. Seems like you'll be serving it from an HTTPS site somewhere on the web and also plan to have a the device connected to Stratux at the same time? Will the PWA mostly live in a local cache on the viewing device?

femski, 2019-08-20:
@cyoung 

Sorry it took me some time do all my testing w/ stratux. 

To your question: PWA is served via https from its own domain (say fly.avnav.com - not public and not fully decided yet). Even when phone/tablet goes offline - App works thanks to service worker which serves entire App (w/ fonts, glyphs and everything). So its like any App - works in cockpit offline. We distribute updates via Service Worker and that woks too. I have tested it working in Safari which is the hardest one to get to work. 

Now, https loaded App can not talk to anything but https websocket on iOS Safari and there lies the issue.

To connect to stratux via ssl/https: we registered a subdomain (stratux.avnav.com) mapped to stratux IP address. This works - as long as you are connected to Stratux running dnsmasq DNS server and serve the content using given certificate. We get gdl90 sream over secure websocket (wss) which works in Safari.

Problem with this approach is we make the Certificate public (Startux is an open box) - our certificate will be compromised. Same will happen if Stratux were to get a domain like pi.stratux.me and make it available for the community - so this really does not work. 

I tried Self-signed certificate - but unfortunately in-spite of all the complications - Safari on iOS does not take Self-signed certificate for WebSockets. So that does not work either.

Unfortunate that SSL for iOT is so horribly broken - its not Stratux fault.

I am going to close this ticket for now - as we get closer to launch (another month or so) I would research other options an come back to Stratux community. Thank you.

closed 2019-08-20 by femski

cyoung, 2019-08-20:
Is there a way you can re-base the content to the Stratux URLs (non-SSL)? Maybe serve an index from the Stratux webserver?