cyoung/stratux #791: Issues with V3 SDR

185Tom, 2019-09-13:
1. Stratux version:1.5b2
2. Stratux config:

    SDR
      - [ ] single
      -[ x] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [X ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: Foreflight 11.7.1

    EFB platform: (e.g., iOS 12.4.1)

    EFB hardware: (e.g., iPad Mini 3)

4. Description of your issue:
I just installed a new UAT SDR V3 and noticed two things.
It does not report tower signal levels as the older version did and also it comes online just after the Stratux had finished booting not waiting the 1:30 that the previous version did.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2019-09-13:
Known bug. RSSI values are sent from the radio now, they need to be interpreted properly in `processRadioMessage()`.

https://github.com/cyoung/stratux/blob/master/main/lowpower_uat.go#L117-L121

There's no startup delay on the UATRadio because it isn't a large power consumer like the other SDRs.

185Tom, 2019-09-14:
Thank you for the reply.
I am not quite up to speed how to alter the code. Will there be a published updated code out soon?
Thanks again,Tom

185Tom, 2019-09-14:
Is there a website I can go too get up to speed a bit on modifying the code?

Thanks for any help,

Tom