cyoung/stratux #796: Ozzmaker BerryGPS-IMU V3 compatibility

javedrahman, 2019-11-21:
1. Stratux version: 1.6r1
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source: Anker 10000mAH

    usb cable: Anker

3. EFB app and version: (e.g., WingX Pro7 8.6.2) SkyDemon

    EFB platform: (e.g., iOS 9.2) Android

    EFB hardware: (e.g., iPad Mini 2) Hudl2

4. Description of your issue:
Is the [Ozzmaker BerryGPS-IMU V3](https://thepihut.com/products/ozzmaker-berrygps-imu-v3) compatible with stratux? It seems the have all the right sensors and will free up a USB port as the GPS chip is also on board. I fly in the UK and would like to add an additional SDR for FLARM. This is supported in the EU version of stratux. At the same time I dont want to lose the UAT antenna as I sometimes fly in the US as well. I have something else running off the 4th USB post so would really help to free up one.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2019-11-21:
The IMU is LSM9DS1 which isn't supported, so that wouldn't work.

If you just want to move the GPS, you could get a module [like this](https://www.amazon.com/Navigation-Raspberry-Betaflight-Geekstory-Shipping/dp/B07PRDY6DS) 
 (search around for BN-220 if the link doesn't work) that you could glue to the case and solder to AHRS pins for power and plug into the RPi serial port.

Theoretically it should work OK, but we haven't tested GPS via the RPi serial port in a long time (I think even since moving to the RPi3). It would take some effort on your part to get working for sure, but definitely less than using an unsupported IMU.

VirusPilot, 2019-11-24:
Here is a GPS board that could work for you but as Chris said, the serial port has not been tested for a while. Furthermore at least a baro chip is recommended (BMP280) which you may have to add on top:

[https://store.uputronics.com/index.php?route=product/product&path=60_64&product_id=81](url)

For SkyDemon you don't need the AHRS, right?

PS: maybe I will do some testing during winter, your request is actually a good one