cyoung/stratux #798: 1090 sdr not responding unless 978 sdr is off

germainejacky, 2019-12-10:
1. Stratux version:1.4r5(991f48eaa5)
2. Stratux config: 
    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: wall wart

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: 
I took a new unit to a Ads-b tower yesterday; 978 sdr worked great...plenty of data and planes.  1090 did not work in any configuration.  Both SDR's "on".  
Today, at home, 20 miles from tower. neither sdr worked until I shut the 978 receiver off.  Then the 1090 worked like gang busters. I'm now looking at a cluster of 8 contacts....Hmmmm

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2019-12-13:
@germainejacky - did you unplug the 978? Or turned it off in the interface? You have a green 1090 SDR or blue?

germainejacky, 2019-12-13:
Hi Mr Young and thanks for getting back to me...also thanks for putting this whole shooting match together.  I did unplug the 978...no change.  My 1090 is green.   BUT! This morning I upgraded my firmware from 1.4 to 1.6 on a 32 GB SD card and wa-la!, 1090 is now working like gang busters with the 978 installed and turned on.   Tomorrow I'm going to drive back back out to the tower to test the 978 again but I'm fairly sure it'll work fine.

closed 2019-12-13 by germainejacky

germainejacky, 2019-12-13:
Hi Mr Young and thanks for getting back to me...also thanks for putting this whole shooting match together. I did unplug the 978...no change. My 1090 is green. BUT! This morning I upgraded my firmware from 1.4 to 1.6 on a 32 GB SD card and wa-la!, 1090 is now working like gang busters with the 978 installed and turned on. Tomorrow I'm going to drive back out to the tower to test the 978 again but I'm fairly sure it'll work fine.