cyoung/stratux #800: "SDR, AHRS and GPS Logs"

ashwinbangalore, 2019-12-30:
1. Stratux version: 1.61
2. Stratux config:
    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: VK-172 dongle, connected via extension cable.

    AHRS
      - [X] yes
      - [ ] no

    power source: Anker 10000mAh brick
    usb cable: ?

3. EFB app and version:  FltPlan Go
    EFB platform: Android 5.5
    EFB hardware: Kindle Fire HD 8

4. Description of your issue:
I'm trying to look at the GPS and AHRS logs by clicking on the link in the "Logs" menu.
It takes me to http://192.168.10.1/view_logs, but the page is empty.
Also, the instructions say "Enable device logging from the Settings page", but there is no option to enable device logging.

Please clarify:
1.   If there should be a link to enable the logs in the settings page
2.  If the view_logs page has an issue
3.  If neither of the above, how do I view the device logs?

FYI, I'm using a FireFox browser on the Kindle Fire tablet.
This issue is reproducible from a laptop as well.

Thank you.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2019-12-31:
Tap the version number at the top of the settings screen numerous times (like 10) to put your Stratux into developer mode. That will then give you an additional settings page to enable logging. This was hidden because people would turn logging on not knowing what they were doing and fill up their SD card and then wonder why their Stratux stopped working.

ashwinbangalore, 2019-12-31:
Thank you so much!

Is there any log-rotate type of service I need to enable that keeps
compressing and deleting prior days' logs?

pp4life, 2019-12-31:
> Is there any log-rotate type of service I need to enable that keeps
compressing and deleting prior days' logs?

That I'm not sure of. I seem to remember them changing how the logs worked at one point. But I can't remember if they made it automatic like that. If you go back through the release notes, you might be able to find what the change was.

ashwinbangalore, 2019-12-31:
Ok. Is there a way to turn off the developer mode? I clicked the version 10
times but nothing happened.

Thank you.

pp4life, 2019-12-31:
Nope, no way to turn it off except reimaging the SD card. But leaving it on doesn't do anything except expose the extra setting menus in the web interface. I just leave it in developer mode.

ashwinbangalore, 2019-12-31:
Makes sense. Thank you so much!
Really appreciate all your help.

closed 2020-01-02 by ashwinbangalore