cyoung/stratux #807: Testing Stratux Integration

leearmstrong, 2020-02-20:
Hello,

We are looking at parsing GDL90 in an application as we have some Stratux users.

One area of interest to us is testing the weather/NEXRAD products.

Is there a way to get Stratux to send the test data that is in the GDL90 documents at all?  We are based in the UK and so don't see any 978 here!

The 2 test APDU frames in the Garmin docs are... Is there a way to get Stratux to send these out repeatedly as a test?

```
 testFrame := ""
if id == 0 {
      testFrame = "130000FC000084A570308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108208000FC000084A3AE00090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084A1EC00090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930130000FC000084AAB7308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108208000FC000084A8F500090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084A73300090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930130000FC000084AFFD308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
} else {
      testFrame = "208000FC000084AE3B00090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084AC7900090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930040000FC000004B1BDF0040000FC000004AFFBD0040000FC000004AE39D0040000FC000004AC77D0040000FC000004AAB5D0040000FC000004A8F3D0040000FC000004A731D0040000FC000004A56FE0040000FC000004A3ADE0040000FC000004A1EBE000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
	}

```

cyoung, 2020-02-20:
Hi Lee,

What Garmin document is that from?

Those appear to be missing the first 8 bytes which contain some of the ground station metadata (lat/long, etc). Copy/pasting the header from an actual tower capture, the first `testFrame` decodes and contains 7 regional NEXRAD frames, and the second `testFrame` contains another 12 regional NEXRAD frames. Here's how you can get stratux to send the data over GDL90.

1. `service stratux stop`.
2. `/usr/bin/gen_gdl90 -uatin`. You'll log output on the console. Stdin accepts UAT packets in dump978 format (see below).
3. Paste in the UAT packets as you want them to be transmitted over the GDL90 stream. You should see the stats in your stratux webui update as if it received something over the radio (http://192.168.10.1/).

Header used: **3C2643887CDCAB80** - a tower at (42.2926, -84.0321).

```
testFrame1:  +3C2643887CDCAB80130000FC000084A570308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108208000FC000084A3AE00090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084A1EC00090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930130000FC000084AAB7308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108208000FC000084A8F500090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084A73300090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930130000FC000084AFFD308950111A53120930110A23451B0A0918090A1B0C1D0607061D041B0A0108000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000;

testFrame2: +3C2643887CDCAB80208000FC000084AE3B00090A1314150617061D04130A01080112131C0D06270615140B0A01000112131C0D06270615140B0A010000090A1314150617061D04130A0108148000FC000084AC7900090A1B0C1D0607061D041B0A010808110A23451B0A091018111A53120920308930040000FC000004B1BDF0040000FC000004AFFBD0040000FC000004AE39D0040000FC000004AC77D0040000FC000004AAB5D0040000FC000004A8F3D0040000FC000004A731D0040000FC000004A56FE0040000FC000004A3ADE0040000FC000004A1EBE000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000;

```

leearmstrong, 2020-02-20:
Ah amazing. I’d missed that part!!

I was referring to this document BTW. 

https://www.faa.gov/nextgen/programs/adsb/archival/media/gdl90_public_icd_reva.pdf

Thanks for coming back to me!

leearmstrong, 2020-02-20:
This worked great, thank you!

closed 2020-02-20 by leearmstrong