cyoung/stratux #810: Fan Control PWM issues

rlsgit, 2020-03-24:
1. Stratux version: v1.6r1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ X] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Fan control may not like certain fans. I have tested mine (Easycargo cooler kit from amazon) and it doesnt like low PWM values. Looks like min of 5-7 for spin.

Can we add a config file to fan control for simple on (10) off (0) control of fan? 

I will work on the code now, but may be easier for a current dev.

VirusPilot, 2020-03-24:
A while ago I did exactly what you described, see attached fancontrol.go (for uploading I had to rename it to fancontrol.txt). It switches the fan to 100% above 50 degrees, and off below 45. You can certainly modify this according to you own needs.

[fancontrol.txt](https://github.com/cyoung/stratux/files/4377054/fancontrol.txt)

rlsgit, 2020-03-25:
i just did that as well, though simply set the pwmminduty to 8.

I may make it more configurable. and submit as a proposed enhancement...

I've found building stratux to be hit or miss...the dependencies arent very obvious.

cyoung, 2020-08-19:
1. If the fan doesn't spin up at a low PWM level, all else held equal, the temperature will continue to be above the setpoint (50ºC).
2. If the temperature is above the setpoint, the PWM level is increased gradually until it reaches "full on".
3. If the failsafe temperature (65ºC) is ever reached, the fan is turned on 100% and fancontrol stops attempting to control the fan using PWM.

These things together mean that the fan will eventually spin up. If the fan doesn't respond well to low duty PWM, then it just will take more time to spin up.

closed 2020-08-19 by cyoung