cyoung/stratux #811: Use actively maintained dump1090-fa.

Determinant, 2020-03-28:
By reading the patch that was made to dump1090, I'm able to reimplement the existing stratux support on dump1090 implementation by FlightAware -- it is a successor to and the most actively maintained version of the original mutability/dump1090.

I also tested the implementation on my stratux build and it works flawlessly. Would it be a great idea to use this cleaner implementation that is based on the latest version of dump1090?

See: https://github.com/flightaware/dump1090/pull/61

VirusPilot, 2020-03-29:
Ted, great idea, just tested it with our Stratux Europe fork [https://github.com/b3nn0/stratux](url) and it works fine so far. I have been working in parallel on some other modification in our Europe fork for Pi4 compatibility which actually complement your work and also work with Pi3:
- replace outdated jpoirier/librtlsdr github version by apt version to stay current (see https://osmocom.org/projects/rtl-sdr/)
- replace outdated wiringpi github version by http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/

Determinant, 2020-03-29:
I'm glad that you like the idea! I also built my own up-to-date wiringpi version with a patched Makefile that allows building the static library for my own Pi Zero build: https://github.com/Determinant/stratux-zero/blob/master/wiringPi_Makefile.patch

I really hope these dependencies could be more actively maintained and apparently there are enthusiasts like us who are willing to contribute.

Determinant, 2020-03-30:
According to dump1090 upstream requirement, the default dump1090 will not listen at port 30006 for stratux (due to historical reason). So to enable stratux support in the latest PR, please use `--net-stratux-port 30006` when `/usr/bin/dump1090` is invoked.

VirusPilot, 2020-03-30:
> According to dump1090 upstream requirement, the default dump1090 will not listen at port 30006 for stratux (due to historical reason). So to enable stratux support in the latest PR, please use `--net-stratux-port 30006` when `/usr/bin/dump1090` is invoked.

So where do I need to add this exactly in my stratux code?

Determinant, 2020-03-30:
There are two ways to do it:

A bit more intrusive way: modify sdr.go to add '--net-stratux-port', '30006' to the line which starts the binary (should be very easy to locate).
A way without changing stratux at all: put the following wrapper script in place of /usr/bin/dump1090:
```
#!/bin/bash
/opt/stratux/stratux_src/dump1090/dump1090 "$@" --net-stratux-port 30006
```
change `/opt/stratux/stratux_src/dump1090/dump1090` to the location of your compiled dump1090 and `chmod +x /usr/bin/dump1090`

VirusPilot, 2020-03-30:
Implemented the latter one - works as expected.

cyoung, 2020-08-19:
@Determinant - will you make those changes to get flightaware/dump1090#61 merged? We can switch over to that.

Determinant, 2020-08-19:
@cyoung Yeah. I'll try to find some time to make changes as required. It will be merged soon after I finish them.

Determinant, 2020-09-08:
The PR has already been merged into the dev branch of flightaware/dump1090. Will be available in the release soon.

Determinant, 2020-11-05:
The latest release of dump1090 (4.) already has support for Stratux: https://github.com/flightaware/dump1090/blob/master/debian/changelog#L7

spineaker-actual, 2021-05-30:
> There are two ways to do it:
> 
> * a bit more intrusive way: modify sdr.go to add '--net-stratux-port', '30006' to the line which starts the binary (should be very easy to locate).
> * a way without changing stratux at all: put the following wrapper script in place of /usr/bin/dump1090:
>   ```
>   #!/bin/bash
>   /opt/stratux/stratux_src/dump1090/dump1090 "$@" --net-stratux-port 30006
>   ```
> 
> change `/opt/stratux/stratux_src/dump1090/dump1090` to the location of your compiled dump1090 and `chmod +x /usr/bin/dump1090`
> 
> I would recommend the latter for now because it doesn't need to change the existing code, while it gives a way to test out any possible dump1090 implementation on the fly.

Hi, would you mind posting a slightly more n00b friendly version of this guidance? I have the most recent image: v1.6r1-eu023-us 
- I would like to leverage dump1090-fa as you are describing, without waiting for it to be incorporated into StratuX
- I'm getting lost on the second "cleaner" method because the image doesn't have anything in` /opt/ `and I'm not quite following the guidance.
Thanks!

Determinant, 2022-02-11:
@cyoung Bumping this issue as the required change was already merged a while back...

wx4cb, 2022-10-26:
im wondering how easy it would be to implement hackrf one support (libhackrf) as i mentioned it here.

we could then implement (using the built in transmitter of the hackrf) adsb-out potentially or even aprs as suggested here https://github.com/cyoung/stratux/issues/836

https://github.com/cyoung/stratux/issues/864

kdknigga, 2022-10-26:
ADS-B out cannot legally be from a portable device.

rhole, 2022-10-26:
I believe you will find also that all transmitters must be FCC type accepted.

 

Rick Hole

wx4cb, 2022-10-26:
>ADS-B out cannot legally be from a portable device.
 
@kdknigga interesting, i didnt know that.

wx4cb, 2022-10-26:
> I believe you will find also that all transmitters must be FCC type accepted. 

i can see that for something like adsb out etc, but that said, if you are doing it under your amateur radio license you're ok as long as you're on an amateur radio frequency you are licensed for (or the ISM band)