cyoung/stratux #813: Aircraft Position on UAT 978 Enhancement

quinntm, 2020-07-01:
1. Stratux version: 1.6r1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ ] yes
      - [X ] no
      type:

    AHRS
      - [X ] yes
      - [ ] no

    power source: Battery

    usb cable:

3. EFB app and version: Foreflight v12.5

    EFB platform: IOS 13.5.1

    EFB hardware: Ipad

4. Description of your issue:

In the United States a large portion of the general aviation community have elected to use the low cost solution of using the 978 band for ads-b. This includes me. I noticed that available aircraft positions on the 978 band are not being displayed on the stratux traffic page, thus not being sent to the EFB.

I was wondering if there are any thoughts on parsing aircraft positions on 978 making them available to display? It seems that Flightaware has done some of this work and I hoping that it could be included to improve traffic awareness in the cockpit.

Excellent product and I rely on it on every flight. It definitely has made a difference of the safety of the aviation communtiy.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

kdknigga, 2020-07-01:
Are you seeing this lack of 978 MHz UAT traffic on the ground, or in the
air?

quinntm, 2020-07-01:
Hi Kris,

I just did some further experiments and everything is all good. I just set
up my receiver by the window and just so happen to have a C152 fly over
transmitting on 978. All is well and you can cancel out the issue request.
Sorry for generating something that was already completed. Thanks for the
great product.

closed 2020-07-04 by quinntm