cyoung/stratux #819: Which receiver is which

dschier, 2020-08-29:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (Foreflight)

    EFB platform: (iOS 13.6)

    EFB hardware: (new iPad 128GB Cellular)

4. Description of your issue: I have two homemade J-Pole antennas mounted in the nose of my Cozy IV.  I recently got a NanoVNA and I would like to tune the antennas for 978 and 1090 but I do not know how to tell which Stratux receiver is which.  There must be an easy way to tell.  Thanks

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

wa9als, 2020-08-29:
The newest UAT radio has a red PCB and is a little larger than the
previous. If you don’t have it, you owe it to yourself to get it. Much more
sensitive for towers in my area.

dschier, 2020-08-29:
Hi WA9ALS, K1UHF here.  I just checked one of the PCB's in my airplane's Stratux and it is no red, but tan.  Now I have to find red board ones. Thanks, 73

I was just testing my ADS-B in J-Pole antennas that I made with a nanoVNA, great device.  They tested < 2:1 VSWR so I put the covers back on the plane; doubt I would notice the difference on towers received if I tuned them better
[image]
[image]

.

dschier, 2020-09-04:
I got new Nooelec NESDR SMArt in aluminum cases. My old ones were blue plastic with tan PCBs.  I hope these are better, if I get a chance to fly with them and they don’t seem to work better Amazon will take them back.

73

Del, K1UHF

Cozy IV N197DL