cyoung/stratux #820: What is AHRS2 beta support?

topnotchaero, 2020-09-21:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Helno, 2020-09-22:
The MPU9250 used in the first generation AHRS board is an obsolete part so the newer boards are using a current chipset the ICM-20948. 

The change in hardware requires a software update. This is that software update.

wa9als, 2020-09-22:
Same functionality?

topnotchaero, 2020-09-22:
If I purchased mine on 11.29.17, is it obsolete?  Do I need the new board?

 

____________________________________

 

TOPNOTCH AERO, INC.

Michael Sawyer

TopnotchAero@gmail.com <mailto:TopnotchAero@gmail.com> 

512.708.8000

 

"The master in the art of living makes little distinction between his work and his play, his labor and his leisure, his mind and his body, his information and his recreation, his life and his religion. He hardly knows which is which. He simply pursues his vision of excellence at whatever he does, leaving others to decide whether he is working or playing. To him, he's always doing both."   James Michener

 

 

 

===================================================================================
CONFIDENTIALITY NOTICE: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, you may not review, copy or distribute this message. If you have received this email in error, please notify the sender immediately and delete the original message. Neither the sender nor the company for which he or she works accepts any liability for any damage caused by any virus transmitted by this email.
===================================================================================

Helno, 2020-09-22:
This is a manufacturing update to allow continued production it does not change the base functionality.

closed 2020-09-22 by Helno