cyoung/stratux #821: Can BCM2385 (Rpi Zero W) provide enough processing power?

snullp, 2020-11-14:
1. Stratux version: v1.5b2
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [ ] yes
      - [x] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source: Power bank

    usb cable:

3. EFB app and version: N/A

    EFB platform: N/A

    EFB hardware: N/A

4. Description of your issue:

I am trying to install Stratux on Raspberry Pi Zero W with the usb hub extension. I learned that its Soc BCM2385 is a single core chip used in Rpi 1. Although I'm attracted by its ultra small form-factor, I don't want the processing power be the bottleneck.

I understand the Stratux data processing is mostly written in Go, which should be pretty light weight. However I'm not sure how the webserver is implemented, and how much processing power that consumes.

Any experiences, comments or suggestions?

pp4life, 2020-11-14:
It has enough processing power. The issue is that it generates a large amount of EMI and makes it so the GPS has a hard time getting a fix and the sdrs perform poorly. Search on Reddit.

snullp, 2020-11-14:
Thank you pp4life. These EMI thingy are interesting and I am very much curious. I now decided to keep current sdrs on rpi3b and buy another set of sdrs for the rpi0, and compare them after I wrap the rpi0 w some copper foils.

matthunter1448, 2020-11-23:
> Thank you pp4life. These EMI thingy are interesting and I am very much curious. I now decided to keep current sdrs on rpi3b and buy another set of sdrs for the rpi0, and compare them after I wrap the rpi0 w some copper foils.

Where are you getting the fireware for the pi0  i'm trying to do the same thing

snullp, 2020-11-24:
I am following https://github.com/Determinant/stratux-zero