cyoung/stratux #835: Option to disable/enable AHRS

pasniak, 2021-04-12:
Is it possible to disable AHRS in the GUI? or via SSH? (I want to avoid Stratux disassembly/reassembly). My device receives many AHRS messages (76). Maybe it would lower power consumption and WiFi traffic, too, if AHRS is not used?

1. Stratux version: 1.5b1

cyoung, 2021-04-12:
http://192.168.10.1/ -> Settings -> AHRS Sensor.

Last we checked the power consumption from AHRS and its related processes was negligible. If you measure it and notice more than a 5% difference, please re-open the issue and note your findings here.

closed 2021-04-12 by cyoung

pasniak, 2021-04-17:
I upgraded to 1.6r1 but I see no "AHRS Sensor" button anywhere... 
- in GPS/AHRS there is only Set leve/zero drift
- in Settings there is only Set AHRS Sensor Orientation and G Limits

cyoung, 2021-04-18:
Ah, sorry. Enable developer mode by clicking 8 times within 3 seconds on the version number on the main status page (http://192.168.10.1/). Then check.