cyoung/stratux #836: Feature: APRS Integration - Discussion

ooglek, 2021-04-25:
I am still new to APRS, and have been a ham radio Technician for 5 years. I open this issue as a long-term discussion on the pros and cons of such an integration and whether it meets the goals of the Stratux development and mission.

This would be an optional feature. Whether the software required comes standard or is added after the fact can be determined later.

**Goals**

1. Connect a ham radio (HT, Handheld Transceiver) audio connection to Direwolf such that incoming APRS messages are decoded as location data and presented to the Stratux user as an _optional_ point on the map with Call Sign.
2. Send the Stratux GPS Location via APRS via the connected HT at some regular configured interval; HT setup depends on VOX or PTT ability via software/direwolf
3. _maybe_ Send the current Stratux GPS location to aprs.fi. This would require the user to modify their install of Stratux to utilize [an external hotspot](https://www.reddit.com/r/stratux/comments/70pmmt/guide_turn_your_stratux_into_a_internet/) such that the client could access both the Stratux on 192.168.10.1 as well as the rest of the internet.

**Challenges and Requirements**

There are a few barriers to entry here:

1. Requires an amateur radio license in the US (take a test, 10-year license; studying takes some effort)
2. Requires an amateur radio audio connection to the Stratux (cost should be low, $25-$50 for starter radio)
3. Requires a physical audio connection via 1/8" audio jack to Stratux, or a Bluetooth Audio Connection (TBD)
4. Requires a dedicated HT to be connected to Stratux; the pilot/operator would need to use a separate HT for audio communication (AFAIK, to be confirmed)

**Discussion**

APRS allows ham radio operators to broadcast their location using 1200 baud audio on 144.39 MHz in the US. An amateur radio license in the US is required to broadcast on this 2-meter (2M) frequency.

Linux-based OSS, such as [Direwolf](https://github.com/wb2osz/direwolf) and [xastir](https://xastir.org/index.php/Main_Page) (HTTPS broken, sad), exists to make encoding/decoding (MODEM!) of the audio stream and location tracking easy. (Does it take AT commands?!? Oh the 80s...)

Audio would be connected to the Raspberry Pi via audio cable or bluetooth audio for input/output. HTs without bluetooth could use an external Bluetooth audio device, compatibility TBD. Challenges include triggering APRS out via VOX and/or PTT to the connected HT.

In the US, Ultralights, FAR Part 103 aircraft, and other aircraft that do not have electrical systems are exempt and excluded from ADS-B out. Using a Stratux and APRS, it would give the flying community a way to broadcast location and potentially avoid undesirable interactions in the air.

APRS support would allow location data to be shared without requiring a terrestrial-based cell network in order to send and receive location data while in flight or away from civilization. 

**Why APRS?**

1. Ability to have fellow pilots see your position on their own map without relying on terrestrial based systems. This would be helpful when flying outside of a cell network range and with aircraft not required to broadcast ADS-B Out
2. Augments Stratux with the ability to Broadcast location to others nearby, which could help ultralights that are not yet allowed (in the US) to broadcast ADS-B to show up on a map somewhere.
3. Emergency way to broadcast location where no cell network exists.
4. Only external hardware and a bit of new software on the Stratux is required
5. Get a different set of people interested in amateur radio and actually get licensed
6. It'd be cool to integrate flying a paramotor with amateur radio. Think of the contacts you could get while 3 miles in the sky!
7. It's a potentially fun challenge 

I will help build a prototype as I learn more about APRS, how it works, what is needed to make it work, and how it could, easily and with little-to-no hardware changes on the Stratux, integrate into the hardware. I don't love the physical connection, so ideally there's a bluetooth audio adapter that could eliminate the cord between the amateur radio and the Stratux onboard Bluetooth that will send/receive the audio without issue.

saabnut, 2021-07-27:
This is interesting, Peter. I assume you are aware of FLARM, which is used in gliders. I feel that Stratux would not be able to perform the request as it is already using 2 USB ports for 2 SDRs, and I would expect a 3rd SDR would be required, for the additional frequency. How often can APRS update position within current HAM license rules. Certainly there is a use case here, but I see it as a separate product. Given all of that, I would go with PowerFlarm. https://www.skybrary.aero/index.php/FLARM  https://flarm.com/products/powerflarm/product-selector/   and we already have Flarm ready stratux code. After you look at this stuff long enough, the FAA may have picked the wrong horse.

wx4cb, 2022-10-26:
maybe using the transmitter side if we can get hackrf-one support implemented we could use that for aprs out. doesn't need to have high power due to altitude :D