cyoung/stratux #837: ForeFlight "GDL 90 Extended Specification" - add new features

cyoung, 2021-04-29:
Broadcast message seems useful. See [here](https://www.foreflight.com/connect/spec/).

Any interest?

spineaker-actual, 2021-05-04:
Yes, when using an aircraft with multiple services on a wifi network, it would be very helpful to integrate a StratuX via ethernet so the EFB can be on one WiFi for all services and still access the StratuX....I think this would help since it wouldn't be a WIFI client....kind of like adding a network IP destination automatically in the settings web gui page?

cyoung, 2021-05-04:
@spineaker-actual - yes, exactly. It would enable stratux to be a client on a network and automatically detect ForeFlight (or I guess other ADS-B consuming devices too, if they adopt this standard).

jamez70, 2021-05-21:
Would the static IP assignments work for this?