cyoung/stratux #842: TTGO T-Beam GPS problem

Dolqus, 2021-09-26:
1. Stratux version: 1.6r1-eu026
2. Stratux config: default

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: TTGO T-Beam v1.1 ublox NEO-M8N

    AHRS
      - [x] yes
      - [ ] no

    power source: 3A 5V power supply

    usb cable: tested with two different cables (which both work)

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
not relevant
    EFB platform: (e.g., iOS 9.2)
not relevant
    EFB hardware: (e.g., iPad Mini 2)
not relevant
4. Description of your issue:
Stratux does not receive TTGO T-Beam GPS NMEA signals, at the website is written "GPS hardware: Serial port (not communicating).
I also tried it with a different raspberry pi, nothing connected but the T-Beam and the same issue appears.
T-Beam GPS tested and working.
The T-Beam connects as serial device on ttyACM0 but it has a baud rate of 115200, so the NMEA text is only shown with
cat /dev/ttyACM0 after doing stty -F /dev/ttyACM0 115200.
Is there a way to tell stratux to listen at /dev/ttyACM0 with a baud rate of 115200?
(Logs are made on a fresh flashed version)
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.lo
[stratux.log](https://github.com/cyoung/stratux/files/7231396/stratux.log)
[syslog.txt](https://github.com/cyoung/stratux/files/7231398/syslog.txt)
g.

b3nn0, 2021-09-26:
Stratux will autodetect the baud rate. It will also try 115200. "Serial port (not communicating)" would mean that your T-Beam isn't found at all - so Stratux falls back to the PI's UART pins. The Stratux process looks for /dev/serialin, which is symlinked to your T-Beam by udev (/etc/udev/rules.d/10-stratux.conf). Are you missing that file? Does /dev/serialin exist (I'd assume it doesn't)?

You can ignore the Length too short debug message. It will disappear once we have GPS Data.

Dolqus, 2021-09-26:
Hi b3nn0, 
yes, /dev/serialin does not exist and the serial ports are linked this way in /dev/:
serial0 -> ttyAMA0
serial1 -> ttyS0

b3nn0, 2021-09-26:
Does your /etc/udev/rules.d/10-stratux.conf correct (like this: https://github.com/b3nn0/stratux/blob/master/image/10-stratux.rules )? Because that's the one that causes /dev/serialin to exist.
What does lsusb say? Maybe unplug your T-Beam, then run "udevadm monitor" and plug it back in to see more detail on what happens.

Dolqus, 2021-09-26:
yes, 10-stratux.conf does exist and is correct.
lsusb says following:
Bus 001 Device 006: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 004: ID 1a86:55d4 QinHeng Electronics
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast                                                            Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
(the T-Beam is Bus 001 Device 004: ID 1a86:55d4 QinHeng Electronics)
udevadm monitor does say this:
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[884.366054] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (u                                                           sb)
KERNEL[884.366968] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0 (usb)
KERNEL[884.368214] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0/tty/ttyACM0 (tty)
KERNEL[884.368454] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0 (usb)
KERNEL[884.368607] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.1 (usb)
KERNEL[884.368799] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.1 (usb)
KERNEL[884.368961] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (u                                                           sb)
UDEV  [884.394081] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (u                                                           sb)
UDEV  [884.402173] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0 (usb)
UDEV  [884.405475] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.1 (usb)
UDEV  [884.411830] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.1 (usb)
UDEV  [884.418299] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0/tty/ttyACM0 (tty)
UDEV  [884.427497] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-                                                           1.2:1.0 (usb)
UDEV  [884.455548] bind     /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (u                                                           sb)
^C

I did flash the recent Stratux image, first I tried yours but obviously the results are the same.

b3nn0, 2021-09-26:
Interesting.. Apprently TTGO changed the USB/Serial adapter on the T-Beam. It always was a CP2104 - now it is a QinHeng Electronics (never heard of that one before..).

I don't have such a T-Beam, so it's hard for me to test.
Could you
- Enable persistent logging (to make sure that changes to the filesystem are actually stored on disk) and reboot
- Add this line to 10-stratux.conf:
```
SUBSYSTEM=="tty", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="55d4", SYMLINK+="serialin"
```
- Reboot again

and see if it works?

Dolqus, 2021-09-26:
correction: 10-stratux.conf does actually exist as 10-stratux.rules (hope that makes no difference)
I did buy it here: https://www.amazon.de/gp/product/B08KVZD6JK
My T-Beam was produced februrary this year.

Sadly those changes did not work, it looks like the 10-stratux.rules gets resetted every reboot

b3nn0, 2021-09-26:
Then I think you forgot the first point: enable persistent logging in the web interface (or run `overlayctl disable` and reboot).
That's because of the read-only overlay root filesystem and needs to be disabled first. And yes, 10-stratux.rules is correct of course.

Dolqus, 2021-09-26:
yes, my mistake I did not rebooted after I enabled persistent logging (too unpatient :D)
and yes it works!! 
Thank you very much and independent of that thanks for your work, we really appreciate!.

b3nn0, 2021-09-26:
Cool. I'll add the line to the next version. Feel free to close this issue then (I can't since I'm not an admin of this repository)

closed 2021-09-26 by Dolqus