cyoung/stratux #845: Starting with v1.6, PI 2 b does not boot (8 blinks) - maybe re-add support?

pasniak, 2021-11-29:
1. Stratux version: 1.6r1
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source:

    usb cable:

4. Description of your issue:

Starting with v1.6 the Raspberry PI 2 b does not boot (shows rainbow pixel and 8 green blinks). I had to downgrade to 1.5b2 (I wanted to see if my app's UDP connectivity drops were problems on Pi's WiFi).

Given it is hard to order PI boards (I see Pi 4 ETA in Jun 2022...) could you consider backporting WiFi fixes to old boards?

And add to each release notes "will not work with PI 2".

cyoung, 2021-11-29:
> Starting with v1.6 

Are you referring to [this release](https://github.com/cyoung/stratux/releases/tag/v1.6r1)?

The others use a different base image.

pasniak, 2021-11-29:
Aha this is possibly the working one I had before (I blindly burned the latest over the working one....). I tried for sure stratux-v1.6r1-eu025-e23a138f-us.img.zip and stratux-v1.5b2-8f4a52d739.img.zip today (NOT working).

cyoung, 2021-11-29:
I don't know what the base image looks like after that image, maybe @b3nn0 can comment better on that.

Was just guessing it was a backwards compatibility issue with the upgrades that were made after then.

One thing you might try is using that base image and doing .sh updates to get to the codebase you're looking for.

b3nn0, 2021-11-29:
Yes, anything that has eu0?? in the image name will not work on the Pi2 any more, because it uses an arm64 base image, and the Pi2 only has a 32bit CPU. I think the code would still run on 32bit when building manually from scratch, but I don't really want to support that, since Stratux now can do quite an elaborate Wifi setup (Wifi-Direct, AP+Client), which is hard to support when we don't know which Wifi dongle the user will use.

so: If you want to use the Pi2, you will either have to do a manual install if you are familiar with Linux (you can use this as inspiration: https://github.com/VirusPilot/stratux-pi4 )
Or you will have to stay with the older, pre-eu?? images.

VirusPilot, 2021-11-29:
> so: If you want to use the Pi2, you will either have to do a manual install if you are familiar with Linux (you can use this as inspiration: https://github.com/VirusPilot/stratux-pi4 )

I have just updated my script [https://github.com/VirusPilot/stratux-pi4](url) so that the armv6l golang package will be installed in case of such legacy platforms.