cyoung/stratux #847: there is a problem is not small

JoAnyad, 2021-12-29:
1. Stratux version:eu27
2. Stratux config: 978 and 1090 RTL-SDR
YES GPS
NO 978 Mhz (UAT)
YES 1090 Mhz (ADS-B/Mode-S ES)
YES 868 Mhz (OGN)
NO 162 MHz (AIS)
YES AHRS Sensor
YES Baro Sensor

    SDR
      - x ] single
      - [ ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

1. has two SDRs. a 1090 and 978.
2. I start the OGN turns off the GPS usb serial and when the 1090ES is turned on there is no OGN data sending
3. disconnect from GPS
i have been fighting like a pig on the ice for three months but i am very bored with this shit. If something is going to be good like WIFI then something is quite xar. the latest version of eu027 is above and the T-BEAM esp32-ogn-tracker-bin-10 + I flashed everything 2000x DE fos. Can someone tell me what's wrong? Or fuck me up?
[stratux.log.txt](https://github.com/cyoung/stratux/files/7790873/stratux.log.txt)
[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-12-30:
Almost certainly power delivery issue.
- Which model of SDRs are you using?
- Is the PI's red power LED constantly on (if not this indicates too low voltage/insufficient power supply)

Try a different USB power supply and a different cable.
See also: https://github.com/b3nn0/stratux/discussions/137

JoAnyad, 2021-12-30:
Thanks for the quick response.
The PI LED is flashing red, and now both LEDs are flashing continuously.
I replaced the power supply (so far 1A) with 2A and the USB cable between PI and TTGO. There is improvement.
In the settings, if I enable (turn on) 1090 Mhz (ADS-B / Mode-S ES), the OGN switches to disconnect on the status page. If I turn on anything (UAT, OGN, AIS, AHRS, BARO) then the OGN works flawlessly, but if I turn on 1090 Mhz (ADS-B / Mode-S ES) it is disconected.
For example, when switching (978 Mhz (UAT)) it is disconnected for one second, then the OGN switches back on properly.
I'm showing you the pictures.
First image: Fully on all except 1090.
Second: the 1090 is also turned on
[image]
[image]

b3nn0, 2021-12-30:
If the red LED is not constantly on, there is not enough power. So that's always the first thing to make sure.
I think there is a conflict in the configuration of your SDRs. I.e. Stratux wants to use both for the same protocol, which doesn't work. Can you post a log file? And again: which SDR model are you using?

JoAnyad, 2021-12-30:
The LEDs are constantly lit. GPS connection is stable :-)
I am attaching the stratux.log file. I hope you thought so because you're a bunch of it :-) I'm attaching RTL-SDR pictures, I don't know which one.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

b3nn0, 2021-12-30:
That explains it. The red one is the Stratux UAT radio and not an SDR. It can only be used for UAT. For OGN-RX you will need a second RTLSDR (e.g. nooelec nano 2+).

JoAnyad, 2021-12-31:
Thanks again for the help!
I ordered NooElec NESDR Nano 2+.
He will arrive on Jan. 11. I hope this will be good.
I am submitting the Stratux.log file because it was left behind yesterday.
[stratux.log.txt](https://github.com/cyoung/stratux/files/7796154/stratux.log.txt)
Would you like to see if there is any other problem?
Happy New Year!