cyoung/stratux #848: Sporadically stops ignoring own aircraft

azjoe67, 2022-02-07:
1. Stratux version: 1.6r1-eu027
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:  VK-162 G-Mouse

    AHRS
      - [x] yes
      - [ ] no

    power source: 25,800mAh power brick

    usb cable: Stratux Serial Adapter CP2102

3. EFB app and version: AvMap EKP-V 1.13.101R

    EFB platform: EKP-V
 
    EFB hardware: EKP-V connected via RS-232 @ 38,400 baud using the FLARM protocol strings

4. Description of your issue:  All is normal until about 30 - 45 minutes after power on. Then Stratux STOPS ignoring my own aircraft and show it as traffic. I have my HEX code properly inputted in the settings. Sometimes, after a few minutes, it will START ignoring my own aircraft again. Other times it continues to display my plane as traffic right on top of me until I power cycle the Stratux.

I'll gather the logs the next time I go to the hangar. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2022-02-07:
Yes, please provide the log files. It will be logged if the ownship config ignoring is intentional and why

azjoe67, 2022-02-08:
I didn't realize that those logs are disabled by default. I've enabled them and I'll upload them next time I fly as it has happened every time I fly. Might not be until the end of the week, but stay tuned .....

azjoe67, 2022-02-08:
Maybe a stupid question, but how do you enable the logs? I've turned them on in Settings, but they don't appear anywhere. I've tried using Persistent Logs, but that didn't seem to do anything either. stratux.log exists and is viewable but http://192.168.10.1/logs/stratux/ returns a 404 error.

b3nn0, 2022-02-08:
Persistent logging needs to be on, yes. Then they are under /logs iirc. The file I need is indeed stratux.log.

azjoe67, 2022-02-08:
I'm in the US, Arizona specifically. The EKP-V only understands FLARM when it comes to traffic. (I'm a beta tester for them) I've got 868 disabled, 978 & 1090 enabled. So far seems to be working pretty well. The only real issue I've seen is my "shadow plane" issue.
As for the logs, I've got persistent turned on, replay turned on. But when I go into logs, all I see are:
auth.log
btmp
daemon.log
fancontrol.err
fancontrol.log
kern.log
lastlog
messages
private/
stratux.log
syslog
user.log
wtmp

Where's the replay log?

azjoe67, 2022-02-08:
Yes, I have ADS-B In/Out: uAvionix Echo.

azjoe67, 2022-02-27:
I've got a bunch of logs, and it did do it again. How do I upload them? I don't see a way to upload.