cyoung/stratux #850: Unable to log in to Stratux raspi

jsampat2, 2022-03-28:
1. Stratux version: stratux-v1.6r1-eu027-3d07b461-us
2. Stratux config:

    SDR
      - [ ] single
      - [ X] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source: Battery pack

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2) 

4. Description of your issue:
I am using a pre-built kit and trying to log in to the Raspberry pi with Stratux v 1.6 Image flashed with log in credentials "pi" and password "raspberry". As seen in the screen below, typing "pi" results in no change, neither does typing "raspberry". Please help me log in to the Raspberry Pi. I even tried flashing a new SD card with the same v1.6 image, but it led to the same result.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log
[image]
.

pp4life, 2022-03-28:
It sounds like your keyboard is not being recognized if you're typing "pi" right there and nothing is showing up. If you're using a wireless keyboard, try a wired one. If that doesn't work, try searching the broader internet for why typing does not show up at the login prompt on my RPi. Or something like that.

wa9als, 2022-03-28:
Which “v 1.6 image, specifically?

jsampat2, 2022-03-28:
@pp4life it was a faulty cable on the keyboard, thank you! I am further trying to access the .go files on the raspberry pi, could you guide me on how to look for the same? Rn I am only able to view the startup service files. 

@wa9als its v1.6r1-eu027-3d07b461-us.img file

pp4life, 2022-03-28:
I can't help you on the programming side. However, why do you need the .go files from the RPi? Aren't they all right here in this repository? It's an open source project, right?

jsampat2, 2022-03-28:
@pp4life Yes, but I am trying to access them on the RPi machine. I basically want to ultimately access all the positional data logs from a run of stratux and save them as a CSV. Do you have any ideas on how to do the same? I am not the greatest at programming either.

pp4life, 2022-03-29:
You can turn on and off logs web interface. I'm not too familiar with what it'll log, but I think everything. Be warned, it's a lot of data and isn't meant to be used by the casual user. Connect to the Stratux via WiFi then open a web browser to 192.168.10.1. In the menu I think you'll find somewhere to turn on logs. If not, tap on the version number at the top multiple times (7, I think) until developer mode is enabled. In any case, I just searched, "Stratux logging" and found several posts on reddit of people trying to do the same thing.