cyoung/stratux #851: Increase Ownership and GeoOwnership message (id 0x0A and 0x0B) rate to better support Stratux on Garmin Pilot.

jeffdamp-wave, 2022-04-02:
1. Stratux version: 1.6r0
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [x] yes
      - [ ] no

    power source: Any

    usb cable: Yes

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Garmin Pilot 10.7.1

    EFB platform: (e.g., iOS 9.2)  IOS 

    EFB hardware: (e.g., iPad Mini 2) iPad mini

4. Description of your issue:
 The 10.7.1 build of GP supports the Stratux device.  Synthetic Vision is very laggy when used with Stratux.  This is because GP ties AHRS data in synthetic vision to also receiving both ownership messages 0x0A and 0x0B (for altitude and GPS).  Right now ownership messages are being sent at 1Hz.  I suggest upping them to at least 5Hz if not 10Hz.  What I did on my local branch was to read the mySituation.GPSPositionSampleRate to set the timer for the reports in heartBeatSender (using a default/max value when SampleRate is invalid).  My branch has diverged so much, not worth attempting a pull request.  I've already sent a bug report to Garmin on the fact they shouldn't tie the AHRS updates to the ownership reports.  A benefit of increasing the rate of ownership reports will help smooth out the 3D map view too.

cyoung, 2022-04-02:
Pretty sure we're following the GDL90 spec on those two messages, the one Garmin itself wrote.

What AHRS message are they using? `0x65 0x01` is sent at 5Hz.

jeffdamp-wave, 2022-04-02:
They are not using 0x65, 0x01 since that is a ForeFlight specific message. They are using 0x4c, 0x45.  Which means they can use more of the data in SV but currently are not.  I verified this by disabling 0x65,0x1.  I also ported the GDL90 code to FS2FF which allows for easy testing in MSF2020.  increase 0x0A smooths out the AHRS orientation while increasing 0x0B smooths out altitude and GPS positioning.