cyoung/stratux #852: AccessPoint and Station Mode not broadcasting to station network

rudiratlos, 2022-05-25:
I've activated WIFI-mode AP+Client 
If iPad with sky-Map is connected to stratux WLAN (hotspot), I can see traffic on my skyMap. If iPAD connected to regualr wlan (Station mode), skymap can not see any traffic.
It looks like taht stratux app is broadcasting traffic (UDP port 4000) only on hotspot (ap0) wlan and not additionally on regular wlan.

1. Stratux version: v1.6r1-eu027
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

b3nn0, 2022-05-25:
This is because Stratux by itself can not know which other devices are on the same network.
If you contact the web interface of the Stratux once, it will remember that and also provide that device with traffic after a couple of seconds.

rudiratlos, 2022-05-26:
thanks for your info, but it's only true for web browser based clients. I have to connected, my mac and my iPad with sky-map on station mode network. Mac/Browser/startux can see the traffic. iPAD with sky-map can't see the traffic. Only when I connect iPAD to ap0-Network (hotspot) then I can see traffic on iPAD/sky-map