cyoung/stratux #854: how to set ssh password

rudiratlos, 2022-07-01:
How can I set a new password for user pi?
Using the default pwd is a security problem

N129BZ, 2022-07-01:
Since the effective radius of a Stratux WiFi signal is around 30 or 40 FEET and any potential hacker would have to be sitting in your airplane, not really much of a security problem... plus what would someone do if they did hack it?  Fix a bug?

wa9als, 2022-07-02:
I hope nobody think this was MY question!

rudiratlos, 2022-07-03:
I did that:
login as pi, set new pwd with passwd, sudo reboot.
relogin: Old default pwd is active.
Problem with read only fs?
default pwd is a security problem: if you use stratux on your home/work lan and not in a plane.
@wa9als: world is not so under complex as it seems in the first place

VirusPilot, 2022-07-03:
You need to disable read only fs in settings: enable "Persistent logging".

rudiratlos, 2022-07-04:
Thanx,
thats it:
enable "Persistent logging"
reboot
ssh pi@stratux.local (login with old pwd)
passwd

go to stratux-website / settings
disable "Persistent logging"
reboot

ssh pi@stratux.local (login with new pwd)