cyoung/stratux #858: http://stratux.me/ is 404

dgzzhb, 2022-09-01:
http://stratux.me/ is 404

Config issue? And why not https?

VirusPilot, 2022-09-01:
Chris has closed his website, what are you looking for? A lot of information can be found on our Europe Edition fork: https://github.com/b3nn0/stratux.

cyoung, 2022-09-01:
No, it wasn't closed... Sean (Crew Dog) manages it now. The registrar was changed and there was an issue with transitioning the hosting. It should be back before too long.

closed 2022-09-01 by cyoung

crewdogelectronics, 2022-09-16:
DNS records updated with the new registrar. Shows stratux.me domain as being verified on GitHub. Still getting a 404 error. Anyone good with gh-pages? I looked through the FAQ and couldn't figure out why we are still getting the error.

novwhisky, 2022-09-24:
Snapshot from June 25, 2022 available from the Wayback machine in the meantime http://web.archive.org/web/20220625024754/http://stratux.me/

mnordhaus, 2022-12-15:
> DNS records updated with the new registrar. Shows stratux.me domain as being verified on GitHub. Still getting a 404 error. Anyone good with gh-pages? I looked through the FAQ and couldn't figure out why we are still getting the error.

I am good with static websites and have done a few things in that arena (GH Pages). I am happy to help.

saabnut, 2022-12-17:
I noticed that the archived page is http://    I have seen issues like this
before, where pages won't render if not https://   As Microsoxx owns
github, I would not put it past them to fail to direct pages based on
"security".

historically, we operated on git hub this way:
https://stackoverflow.com/questions/40659168/difference-between-gitgithub-com-and-https-github-com

See resolving conflicts with mixed content here:
https://docs.github.com/en/pages/getting-started-with-github-pages/securing-your-github-pages-site-with-https

*The page owner should take a look at the "enforce https" check box and see
it is checked.  If so, unchecking it may cure the issue. *
*It may be that you can no longer uncheck said box. I can't load the page
either. My antivirus flags the site as insecure before even though landing
on the 404 page!*

*Also see *Resolving problems with mixed content

I am thinking that the issue may be here:

Enforcing HTTPS for your GitHub Pages site

   1.

   On GitHub, navigate to your site's repository.
   2.

   Under your repository name, click

-

*Settings*. [image: Repository settings button]
-

In the "Code and automation" section of the sidebar, click
-

* Pages*.
-

Under "GitHub Pages," select *Enforce HTTPS*. [image: Enforce HTTPS
checkbox]
Troubleshooting certificate provisioning ("Certificate not yet created"
error)

When you set or change your custom domain in the Pages settings, an
automatic DNS check begins. This check determines if your DNS settings are
configured to allow GitHub to obtain a certificate automatically. If the
check is successful, GitHub queues a job to request a TLS certificate
from Let's
Encrypt <https://letsencrypt.org/>. On receiving a valid certificate,
GitHub automatically uploads it to the servers that handle TLS termination
for Pages. When this process completes successfully, a check mark is
displayed beside your custom domain name.

The process may take some time. If the process has not completed several
minutes after you clicked *Save*, try clicking *Remove* next to your custom
domain name. Retype the domain name and click *Save* again. This will
cancel and restart the provisioning process.