cyoung/stratux #862: Avare connects but not data

shempferd, 2022-09-20:
1. Stratux version:
2. Stratux config:

    SDR
      - [x ] single
      - [ ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ x] yes
      - [ ] no

    power source:

    usb cable:  I am able to connect to the stratux wifi and I see the data on Avare internal external app,  It just doesnt show on the AVARE app.  No traffic and I cant pull up weather??

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:  

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2022-09-21:
You don't need the Avare separate I/O app anymore. Just connect to the Stratux WiFi, open Avare then on the bottom scroll all the way over to "IO" and set the dropdown at the top to WiFi. Data should start flowing.

shempferd, 2022-09-21:
Got it.  Thanks for that.  

I still am not seeing weather on AVARE nor can I pull up the winds aloft as before.  I am seeing one aircraft but normally see way more.  I haven't updated the firmware in a couple of years.  I see the newest one says Europe.  Will that also work in the USA?  Should I need to update even though I am able to connect my tablet to the Stratux?

Thanks

wa9als, 2022-09-21:
Yes update to the latest eu028 with US settings, and hopefully you have the low power UAT radio with the red PCB - more sensitive for towers.

shempferd, 2022-09-23:
I had to click on the io tab while flying to get weather.  Seems like a bug.