cyoung/stratux #864: Question hackrf one

wx4cb, 2022-10-26:
I tried a quick search, but is the hackrf one sdr supported at all?

I ask as I have one lying around and it could be a good way to have adsb in as well as out as it does have a transmitter module built in.

wx4cb, 2022-10-26:
Maybe it can be replaced with this version of dump1090 - installing now so we'll see

https://github.com/itemir/dump1090_sdrplus

also found this too 
https://medium.com/@rxseger/flight-tracking-with-ads-b-using-dump1090-mutability-and-hackrf-one-on-os-x-3e2d5e192d6f

VirusPilot, 2022-10-27:
All the latest Stratux images support HACKRF in dump1090, only BLADERF ist disabled.

VirusPilot, 2022-10-27:
Strange. I recommend to clone https://github.com/flightaware/dump1090 on a spare Raspberry Pi and start from there.

wx4cb, 2022-10-27:
I tried that but it doesn't have the dashnet - stratics command line
parameter

If I remember right but I will look into it further when I get home later

lyusupov, 2023-02-24:
> I ask as I have one lying around and it could be a good way to have adsb in as well

https://github.com/lyusupov/SoftRF/wiki/ES-Edition