cyoung/stratux #866: Cant install maps from ‘abatzill.de’

nbeli, 2022-12-03:
1. Stratux version: Linux stratux 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64
2. Stratux config: Raspberry Pi 3+

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ ] yes
      - [x ] no
      type:

    AHRS
      - [ ] yes
      - [ x] no

    power source: Ok

    usb cable: Ok

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

When trying to install maps using SSH connection, I get following error message:

BEGIN
pi@stratux:~ $ sudo -s
root@stratux:/home/pi# cd /overlay/robase/opt/stratux/mapdata
root@stratux:/overlay/robase/opt/stratux/mapdata# ./download_mapdata.sh -h

Download OpenFlightMaps VFR Charts Europe (~700 MiB)? [y/n]y--2022-04-05 03:47:09--  https://abatzill.de/stratux/openflightmaps.mbtiles
Resolving abatzill.de (abatzill.de)... 2a03:4000:29:5bb::1, 94.16.117.178
Connecting to abatzill.de (abatzill.de)|2a03:4000:29:5bb::1|:443... connected.
ERROR: The certificate of ‘abatzill.de’ is not trusted.
ERROR: The certificate of ‘abatzill.de’ has expired.
The certificate has not yet been activated
mount: /overlay/robase: mount point is busy.

Download US Sectional VFR Charts (~4.9 GiB)? [y/n]y--2022-04-05 03:47:11--  https://abatzill.de/stratux/vfrsec.mbtiles
Resolving abatzill.de (abatzill.de)... 2a03:4000:29:5bb::1, 94.16.117.178
Connecting to abatzill.de (abatzill.de)|2a03:4000:29:5bb::1|:443... connected.
ERROR: The certificate of ‘abatzill.de’ is not trusted.
ERROR: The certificate of ‘abatzill.de’ has expired.
The certificate has not yet been activated
mount: /overlay/robase: mount point is busy.
root@stratux:/overlay/robase/opt/stratux/mapdata# client_loop: send disconnect: Broken pipe
lmac-nid-dsy:~ nid$ 
END

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2022-12-03:
Already fixed upstream. Wait until you have a GPS fix so the clock is set correctly, or run
 sudo systemctl start systemd-timesyncd
First to set the system time via internet.

nbeli, 2022-12-03:
Thanks a lot Adrian.

nbeli, 2022-12-03:
Hello again,
Download and save went OK but I got a last error, "mount: /overlay/robase: mount point is busy."
Thanks in advance for your assistance.
Best regards.
Nestor

nbeli, 2022-12-04:
Thanks @b3nn0

closed 2022-12-04 by nbeli

nbeli, 2023-02-28:
Hello Adrian,
I had to reflash my Stratux Europe, and I get the same problem again :
mount: /overlay/robase: mount point is busy.
despite my letting the GPS get a fix or executing sudo systemctl start systemd-timesyncd
prior to installing the maps.

Thanks in advance for your support.

Best regards.
Nestor

***@***.***:/overlay/robase/opt/stratux/mapdata# sudo systemctl start systemd-timesyncd
***@***.***:/overlay/robase/opt/stratux/mapdata# ./download_mapdata.sh

Download OpenFlightMaps VFR Charts Europe (~700 MiB)? [y/n]y--2023-02-28 20:40:57--  https://abatzill.de/stratux/openflightmaps.mbtiles
Resolving abatzill.de (abatzill.de)... 2a03:4000:29:5bb::1, 94.16.117.178
Connecting to abatzill.de (abatzill.de)|2a03:4000:29:5bb::1|:443... connected.
HTTP request sent, awaiting response... 304 Not Modified
File ‘openflightmaps.mbtiles’ not modified on server. Omitting download.

mount: /overlay/robase: mount point is busy.

Download US Sectional VFR Charts (~4.9 GiB)? [y/n]nmount: /overlay/robase: mount point is busy.
***@***.***:/overlay/robase/opt/stratux/mapdata#