cyoung/stratux #869: Receiving Flarm in France

gauthier108, 2023-02-28:
1. Stratux version: 
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ X] yes
      - [ ] no

    power source: USB with on board LM2596S, and Raspberry official alim on 220v

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Skydemon 3.16.6.354

    EFB platform: (e.g., iOS 9.2) iOS 16.3.1

    EFB hardware: (e.g., iPad Mini 2) iPhone 13 Pro

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Hello !

I am flying in France, and often in the Alps.
I just build my Stratux (hardware and software) with 2 receivers + stratux-radar-display with OLED Screen.
I receive ADS-B trafic well but I don't receive any glider using Flarm (on the Stratux Webpage and on Skydemon).
I have the OGN option on, but I am not receiving any trafic by this way. I try recently in flight, the stratux did not see any glider that was flying near me, but I could see the glider on the website spot the glider.
Can someone tell me please what I have to do to receive and display glider (using flarm devices), please.

Many thanks

VirusPilot, 2023-02-28:
There could be a couple of reasons why it is not working. Is the antenna or it's connector broken? Is your OGN SDR properly calibrated? Do you have proper GPS reception? As a first step you could look into your logfile to identify potential warning or error messages.

gauthier108, 2023-02-28:
Thanks for your answer @VirusPilot !
The antenna and the connector seems okay.
I configured the 2 SDR receivers with the Sdr-tools, and dedicate each one to one frequency (1090Mhz ADS-B for one, and 868Mhz OGN for the second one). What else should be done to calibrate them ?
At home I have no gps position, but during the flight the reception was OK (3D on the interface and ok also on the OLED display thanks to the stratux radar).
Flying gliders too, I know that we have to update each year the flarms devices by a software available on the flarm website, otherwise gliders will not be seen by the others.
Is the stratux up to date ?
My log file are attached.
Many thanks
[stratux.log](https://github.com/cyoung/stratux/files/10853303/stratux.log)

b3nn0, 2023-02-28:
Hard to tell much without a GPS signal. The OGN Receiver is only started once you have a GPS (as it is required to decode Flarm).

VirusPilot, 2023-02-28:
- the logfile looks fine but remember, at home without a GPS signal, OGN won't start working so maybe (for testing purposes) you can move the GPS antenna somewhere outside but if you don't have gliders flying around you won't see much 
- in your case no further calibration seems to be required
- Stratux needs no yearly update

Further idea: are you sure that you did not mix up the antennas? If the 1090 antenna is connected to the 868 SDR, you won't receive anything (almost).

Btw, you may consider continuing this discussion on Slack where a lot of experts are hanging around to help:https://join.slack.com/t/stratuxadsb/shared_invite/zt-1q8081j50-no9TnQmdLQqvM_6S0ywaNA

gauthier108, 2023-02-28:
Thanks again for your answer !
Okay, normally they are not mixed up : I receive ADS-B trafic without any problems, and in flight ADS-B trafic that are quite far away (up to 40nm I would say).
Will have to check again in flight, I had some SDR deconnexion last flight, probably due to a not enough intensity USB adapter. I will try again with the LM2596.
Okay for the update! 
I will look at the discussion on Slack.
One last question (I search on Github, but did not find a clear procedure), how is it possible to use the stratux to emit my own position on the FLARM frequency with a 3rd SDR ? The idea is to be seen by gliders too.

VirusPilot, 2023-02-28:
One thing to understand is that even with the 868 antenna connected to the 1090 SDR you will receive plenty of ADSB traffic. If you want to send your position via 868 MHz, please look here: https://github.com/b3nn0/stratux/wiki/Build-example-3:-Include-transmission-capabilities,-so-others-can-see-you-as-well. The T-Beam is used by most of us as it comes with a u-blox GPS on board.