cyoung/stratux #871: AHRS - What does "Error identifying IMU: remote I/O error" - mean?

davidkjackson54, 2023-03-10:
1. Stratux version:V1.6r1-eu028
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I have installed a GY-91 and set the pins to the Raspberry PI 3B correctly.
I open up 192.168.1.10 and try to set AHRS orientation but get the error "Orientation failed' 
The log shows:
**Error identifying IMU: remote I.O error**

but I can find nothing to indicate what this actually means and if there is a solution.
Is there any documentation that might help on this error message?

securedirective, 2024-01-30:
Did you ever find a solution? I'm getting this now too. It was working fine a few months ago, and I changed nothing at all. I'm wondering if it's actually a hardware failure, perhaps of the MPU9250 chip.

brightproject, 2024-04-24:
Hello @cyoung and @jpoirier 
A little off-topic, but could you please tell me, `Stratux` uses data from `MEMS` sensors that are already `prepared`, i.e. is sensor fusion done inside the sensor or does the fusion happen in `stratux` code?
For example here
https://github.com/cyoung/stratux/blob/eec0b15d4a820167b8419b4d09b05c85f7896396/main/sensors.go#L168
As far as I understand, Stratux works with two types of sensors `ICM20948` and `MPU9250`, and they seem to contain `DMP` - digital motion processor.
I want to use `lsm6ds3` or `l3gd20` MEMS sensors.
They unfortunately don't do sensor fusion, and so I'm wondering if there is some kind of `magic` in the `stratux` code?
P.S. I program in `C++`, I understand a little `Python`, but I don’t quite understand the `Go` language.