cyoung/stratux #872: Lost Stratux home page on my iPad

mslisaj, 2023-03-30:
I lost the stratux home page on my iPad but the stratux works just fine and drives my EFB perfectly.  I used to open the stratux home page to check the status of the device and also to shut it down.  That page has disappeared and I can't get it back.  Full description of issue is below.

1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ x] dual

    GPS
      - [ x] yes
      - [ ] no
      type:on the board

    AHRS
      - [x ] yes
      - [ ] no

    power source:battery

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
FlyQ
    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:  The stratux works just fine it seems but has lost the home page.  I used to just click the stratux icon on my ipad and I would be taken to the stratux home/status page and I could see what it was doing.  
5.  Here is a side note about this issue.  Apparently I have a wifi extender that has the same address and when I tap the stratux icon the wifi extender opens now.  I took the stratux away from my home and still a partial of the wifi extender page still opens but not the stratux.  The stratux is connected via wifi and it works and drives my FlyQ EFB perfectly but it would be nice to get the stratux home/status page back so I could check the device and give you version numbers and other info you may ask about it.  I sincerely thank you for your help.  I have been using this device for 5 years now and it's been a fun trip and for the last year or so I have had no issue with the stratux but now this.   Again I appreciate your help.
6. best regards,  Lisa
7. I wish I could get to the logs but this is my issue.
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

pp4life, 2023-03-30:
First, connect to your Stratux, then try opening Safari or whatever web browser you have on your device and typing 192.168.10.1 and see if it brings up the Stratux "home page". If not, then search online for something like, "clear Safari web browser cache on iPad". Replace Safari with whatever browser you're using. Another thing to try if you have a second web browser on your iPad: Connect to the Stratux and type 192.168.10.1 into that 2nd web browser and see if it works. 

It mostly sounds like there is some cached data from your wifi extender (or something). 

I use to also use the homepage shortcut, but I stopped using it years ago because I had some weird issue similar to yours. It was so long ago I can't remember exactly what was going. My solution is to have a bookmark in my web browser. It's one extra step to open the web browser first then to click on the bookmark, but it has worked ever since.

pp4life, 2023-03-30:
Glad you got it worked out. Yes, it's a cool project and I've had mine running for 5 or 6 years.