cyoung/stratux #877: Error when executing 'make'

msalmonw, 2023-08-03:
Could someone figure out the issue i'm facing here

root@rock-3a:~/stratux# make
make xdump978 xdump1090 xgen_gdl90 fancontrol
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c anet.c -o anet.o
fatal: No names found, cannot describe anything.
gcc -DMODES_DUMP1090_VERSION=\"\" -O2 -g -Wall -Werror -W  -c interactive.c -o interactive.o
interactive.c: In function 'interactiveShowData':
interactive.c:140:44: error: '%03d' directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]
  140 |                         snprintf(strFl,6,"F%03d",(a->altitude/100));
      |                                            ^~~~
interactive.c:140:42: note: directive argument in the range [-21474836, 21474836]
  140 |                         snprintf(strFl,6,"F%03d",(a->altitude/100));
      |                                          ^~~~~~~
interactive.c:140:25: note: 'snprintf' output between 5 and 11 bytes into a destination of size 6
  140 |                         snprintf(strFl,6,"F%03d",(a->altitude/100));
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:37: interactive.o] Error 1
make[2]: Leaving directory '/root/stratux/dump1090'
make[1]: *** [Makefile:26: xdump1090] Error 2
make[1]: Leaving directory '/root/stratux'
make: *** [Makefile:14: all] Error 2
root@rock-3a:~/stratux#

rvt, 2023-08-05:
Did you clone it like this? `git clone --recursive https://github.com/b3nn0/stratux.git`
Also, this 'might' have something to do with your compiler.

msalmonw, 2023-08-07:
I cloned by following the wiki, using `git clone --recursive https://github.com/cyoung/stratux`
I did not clone the repository by b3nn0. I solved this issue by removing -Werror flag from the Makefile. Now the issue I face is this one:

```
/dump1090/dump1090.h:238: multiple definition of Modes'; dump1090.o:/root/strautx/dump1090/dump1090.h:254: first defined here
collect2: error: ld returned 1 exit status
```

N129BZ, 2023-08-07:
https://github.com/b3nn0/stratux is the current source for both EU and US versions of Stratux and has supplanted the original source.  It fixes numerous issues with the original source and has added many new features, and is eminently buildable on a Stratux-imaged rPi by following the instructions in https://github.com/b3nn0/stratux/wiki/Developing-Stratux, the stratux image includes stuff like convenience scripts and a couple of dependencies that are not in the source code.

If you want to compile on a linux PC, I had to install a couple other things. (tried and succeeded on minimal versions of Debian and Ubuntu, currently trying on Arch)
- go v1.20.7
- build-essential
- cmake
- librtlsdr-dev
- libncurses-dev

msalmonw, 2023-08-07:
Thanks a lot. I have installed these dependencies as well. I'll try the repo at "https://github.com/b3nn0/stratux" now. But, would you be so kind to lay out your own for compiling the code? Including the steps regarding the installation of the dependencies and compilers. It would be really helpful.

N129BZ, 2023-08-07:
Update: Now successfully compiling and running stratux from b3nn0/stratux source on Arch Linux minimal install running in vmware.

N129BZ, 2023-08-07:
further help with getting help to compile stratux source would best be done by joining the Stratux ADS-B channel on Slack

msalmonw, 2023-08-09:
Okay perfect! thanks a lot!

msalmonw, 2023-08-09:
How did you set-up go? I'm getting issues after extracting the go v1.20 tar