cyoung/stratux #878: T-Beam Communication problem

ange12lo, 2023-09-18:
1. Stratux version:  v1.6r1-eu029 
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:
TTGO T-Beam ESP32 868MHz LoRaWAN NEO-M8N GPS Board with SX1262 Chip. 
Fw flashed via Stratux with the command: sudo /opt/stratux/ogn/install-ogntracker-firmware-pi.sh sx1262

lsusb: Bus 001 Device 003: ID 1a86:55d4 QinHeng Electronics USB Single Serial

    AHRS
      - [ ] yes
      - [x] no

    power source:
45w USB-C Power Supply and also 5v 4A Power Source

    usb cable:
tried different Cables

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Get the Error: USB Serial IN (Not communicating) on the Status Page. 
after clicking the RST button on the T-Beam Board, the communication is established. but only until the next power cycle.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

ange12lo, 2023-09-18:
I have found a temporary solution, but it is not satisfactory.

What I have found out so far. 

When I run `cat /dev/ttyACM0' it is empty, so no communication.

The log also says: `serial port/baudrate detection err: Failed to detect GPS serial baudrate'.

Now I have written a script that resets the TTGO T-Beam: `python3 esptool.py --chip esp32 --port /dev/serialin --baud 921600 run` Now it works fine, but what I don't know is why I need to reset the T-Beam. 
Any ideas?

Helno, 2023-10-24:
You should repost this over on the current fork. https://github.com/b3nn0/stratux/issues