cyoung/stratux #879: uAvionix Ping USB support

stefanux2, 2023-10-30:
# uAvionix Ping USB support
## Setup
1. Stratux version: Stratux 1.6r1-eu029 by b3nn0 (EU)
2. Stratux config:

    SDR
      - [X] single using uAvionix Ping USB via USB
      - [ ] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source: 12V to 5V

    usb cable: 10 centimeters

3. EFB app and version: Easy VFR 4

    EFB platform: iOS 17

    EFB hardware: iPad Mini + iPhone 14

4. Description of your issue:
I'm using Stratux built with uAvionix PingUSB device. This patch is to add the MavLink protocol to the stratux in order to enable the USB-Serial PingUSB device.

## Devices
1. Ping EFB
> USB Serial based with 1090 ES Protocol (Currently supported by Stratux)
> https://uavionix.com/downloads/pingefb/uAvionix-pingEFB-ADS-B-receiver-quick-start-guide.pdf
2. Ping USB
> USB Serial based with MavLink Protocol (This patch)
> https://uavionix.com/products/pingusb/

## Protocol
### Traffic Report Message
- Packet Id=**246** from _uAvionix Ping Integration Guide.pdf_
- https://www.google.com/search?q=mavlink+uavionix+integration+guide
[image]

### MavLink
- https://mavlink.io/en/guide/serialization.html#v1_packet_format

## Compilation
### Environment

1. Stratux EU with latest version
2. Remount the Stratux in RW /boot/cmdline.txt
3. apt update and install all the development tools
4. Install Go compiler
5. Create Go Environement
6. Modify main/ping.go
7. Make and copy the executable

### Patch Ping.go
This patch was compiled overwriting the main/**ping.go** file.
[stratux-serial-ping.zip](https://github.com/cyoung/stratux/files/13205377/stratux-serial-ping.zip)

## Features

1. Support uAvionix Ping USB connected via USB to enable MavLink protocol
4. Bearingless target emulation
8. RSSI Emulation

## Test
Tested with 20 hours of flight at sea level up to 8000ft in Europe/Italy.
[image]

## Integration on mainstream
Best integration is to add as second protocol option like "PingEFB" and "PingUSB" switch and rename it as pingusb.go

### Improvements
1. Split ping.go to pingefb.go and pingusb.go
9. Test with multiple disconnect-reconnection of the USB
10. Improve the crosscheck with valid data field
11. Improve Altitude integration with Stratux
12. Improve RSSI for bearingless traffic

cyoung, 2023-10-30:
Are there more than 20 pingUSBs in existence?

alexaggie, 2023-10-30:
I have one….

stefanux2, 2023-11-06:
> Are there more than 20 pingUSBs in existence?

Yes, officially there are up and working more than 100 pingUSBs devices around the world.

GabrielN2002, 2024-01-10:
Could this enable any Mavlink adsb receiver to work with stratux? One of the main problems I am having with my stratux build is the interference caused by the sdr. It basically makes my gps receiver useless. If this works with any Mavlink adsb receiver, like this one: https://www.aerobits.pl/product/aero/, I wouldn't need an sdr and the interference problem could be solved.

stefanux2, 2024-01-12:
**Yes**, I think so, the packet format is the _same_.

A further check shall be:
1. USB Device Id recognised by the system
2. Id of the MavLink Packet shall be confirmed to be 246

If you want to test it, this patch is already merged into the EU version: https://github.com/b3nn0/stratux
Currently up and working on my EU version: https://github.com/b3nn0/stratux/discussions/264

GabrielN2002, 2024-01-17:
From what I have seen on the data sheet provided by Aerobits, it uses the same 246 mavlink packet id. 

The adsb module doesn’t use usb. It connects via uart using a JST GH 6 pin connector commonly found on the pixhawk flight controller. Based on the uavionix rules file https://github.com/b3nn0/stratux/blob/master/image/99-uavionix.rules it seems like the pingusb is using a FTDI uart to usb bridge and it just so happens that I have a JST GH 6P to usb adapter that uses the same FTDI chip. Running the lsusb command in the raspberry pi outputs the same product and vendor id. So in theory it should be recognized by the raspberry pi. 

I’ll update as soon as I receive the ads-b module.

GabrielN2002, 2024-01-26:
Got the ADS-B receiver. So far i haven't been able to get it working. The receiver gets recognized as a PingUSB but the ADS-B is not being received by Stratux. I suspect the problem is the baudrate. The ping.go file has the PingUSB at a baudrate of 57600. The Aerobits Aero uses a baudrate of 115200. I will try changing that.
[image]

GabrielN2002, 2024-01-26:
Changing the baudrate made it work!
[image]

stefanux2, 2024-01-27:
Great, it is time to add this device into the stratux branches...
To differentiate this dongle device:
1) can you set a different 99-uavionix.rules device name https://github.com/b3nn0/stratux/blob/master/image/99-uavionix.rules#L7
2) this allows the ping.go to differentiate the baudrate here: https://github.com/b3nn0/stratux/blob/master/main/ping.go#L57

Can you create the merge request with these mods into the stratux EU/USA branches?